ecocurious2/MultiGeiger #1: Nur kurz ein Verweis auf ein ähnliches Projekt

dadosch, 2019-09-23:
mit SMD-Bauteilen: (für Raspi konzipiert, aber letztenendes kommt auch nur ein Impuls raus)

https://apollo.open-resource.org/lab:pigi:hardware

rexfue, 2019-10-27:
Ja, das kennen wir. Das Board kostet 35€ - das wird im Ganzen zu teuer. Unsere allerersten Gehversuche haben wir mit diesem Board gemacht. Die Schaltung von Jürgen ist einfacher und kleiner.

ThomasWaldmann, 2019-10-27:
Sehr cool dass Euer Bausatz günstiger ist.

Günstiger Bausatz == mehr Potential für Messstellen!

ThomasWaldmann, 2019-11-12:
Ich denke, wir koennen dieses Ticket schliessen, oder?

Was man aber machen / checken koennte (-> neues Ticket), ist eine Differenzierung "MultiGeiger vs. andere" und das in die Docs aufnehmen.

Nordrunner01, 2019-12-26:
Moin.
Na ja, gibt schon einige mehr.
Z.B.: http://www.opengeiger.de/ , viele interessante Informationen zu dem Thema.
Mit dem PiGi habe ich mich auch beschäftigt und die Platine dafür bei PCBs.io verfügbar gemacht. BOM auch erstellt bei octoparts. Einige Teile sind aber wohl nicht verfügbar. :-/

Ansonsten Hut ab für das Projekt! :+1: 

Gruß aus dem hohen Norden

ThomasWaldmann, 2020-02-11:
Denke, wir koennen das Ticket hier zumachen.

Wenn man nach anderen Geiger-Projekten sucht, kann man ja auch ne Suchmaschine nutzen.

closed 2020-02-11 by ThomasWaldmann