ecocurious2/MultiGeiger #3: language?

ThomasWaldmann, 2019-10-19:
the project seems to be a bit undecided whether it should be in English or German, leading to a mix of both in misc. places:

- source code
- user interface
- docs

rexfue, 2019-10-27:
You are right.
I think, docs should be preferably in german because we want also lay persons to build the device.
Source could (should) be in english, only developers would read it ('profis' :-) )
What do you mean with 'user interface' ?

ThomasWaldmann, 2019-10-27:
Actually, I guess the german strings I have seen in the src are for serial (usb) log output.

ThomasWaldmann, 2019-11-03:
In the src, there is quite frequently the GMZ (Geiger-Mueller-Zaehler/Zaehlrohr) abbreviation mixed with english words - this is a rather unpretty.

Better:
- leave it away as we only deal with pulses of that anyway?
- GMC (counter)
- GMT (tube)

ThomasWaldmann, 2019-11-03:
Some of the inconsistencies are fixed by PR #28, but not the GMZ stuff.

ThomasWaldmann, 2019-11-12:
Any comment to my suggestions?

rexfue, 2019-11-12:
I would prefer GMC.  (GMT ist a time zone for me).

ThomasWaldmann, 2019-11-12:
@WwwBoehriDe GMC fine for you, too or do you prefer sth else?

commit ac5dba2d (2019-11-16): Merge pull request #61 from ThomasWaldmann/fix-language-gmc

fix language: GMZ -> GMC, fixes #3
closed 2019-11-16 by ThomasWaldmann