ecocurious2/MultiGeiger #7: 'u8x8_font_7x14_1x2_f' was not declared in this scope

ThomasWaldmann, 2019-10-26:
where do i get this font from?

ThomasWaldmann, 2019-10-27:
Hmm, somehow it works now. Not sure why.

Does one need to restart the arduino ide after installing stuff with the library manager?

rexfue, 2019-10-27:
I don't think so.

ThomasWaldmann, 2019-10-27:
Just a hint, in case someone else sees this error msg from the arduino compiler:

The solution is **not** to change the font names in the multigeiger source code.

I first thought the not declared font is due to a font set change in the library and later that one needs to use `u8g2_font_...` font names now - this does **not** work, do not try that.

Suddenly it worked for me, strange. If you find out why it might not work at first and then works, please add some info here.

ThomasWaldmann, 2019-11-12:
Note: leave this open until 2020-01. If noone can add more information until then (e.g. when / why this happens), I guess we can close it.

ThomasWaldmann, 2019-12-09:
Note: I've never seen the message again since back then.

ThomasWaldmann, 2019-12-18:
@rexfue @WwwBoehriDe did anybody in freiburg workshop have this?

Did they setup their own arduino esp32 environment or did one of you all the flashing?

WwwBoehriDe, 2019-12-19:
The Controller was already pre-flashed. I cant't remember. But I think the one or the other had the environment running on the PC.

rexfue, 2019-12-19:
One of them (Stefan from CCC Freiburg) compiled on his own. He used platformio and had no problems.

ThomasWaldmann, 2019-12-19:
OK, guess we can close this. Not sure how I got that, but it does not seem to be happening often.

closed 2019-12-19 by ThomasWaldmann