ecocurious2/MultiGeiger #8: division by zero bug

ThomasWaldmann, 2019-10-26:
I tested the software from current master branch (plus some changes to get the fonts stuff compiling) only with the esp32 board, without it being pugged into the (not yet assembled) multigeiger board.

Thus, cpm is 0 and it tries to divide by that. Would be nice to avoid that.

```
Gemessen: cpm= 0
SW0: 1  SW1: 1  SW2: 1  SW3: 1
Guru Meditation Error: Core  1 panic'ed (IntegerDivideByZero). Exception was unhandled.
Core 1 register dump:
PC      : 0x400d52d5  PS      : 0x00060430  A0      : 0x800e0ea0  A1      : 0x3ffb1f60  
A2      : 0x00000000  A3      : 0x00000000  A4      : 0x3ffc1564  A5      : 0x3ffbebf8  
A6      : 0x00000000  A7      : 0x00000001  A8      : 0x800d529c  A9      : 0x3ffb1f40  
A10     : 0x0001d4c0  A11     : 0x00000000  A12     : 0x00000000  A13     : 0x3ffb1e90  
A14     : 0x00000000  A15     : 0x3ffb0060  SAR     : 0x0000000a  EXCCAUSE: 0x00000006  
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  

Backtrace: 0x400d52d5:0x3ffb1f60 0x400e0e9d:0x3ffb1fb0 0x4008930d:0x3ffb1fd0

Rebooting...
```

rexfue, 2019-10-27:
OK, will be changed at next version.

ThomasWaldmann, 2019-11-07:
PR #30 says to have fixed this.

ThomasWaldmann, 2019-11-12:
Merged, so hopefully fixed now in master.

ThomasWaldmann, 2019-11-24:
```
Guru Meditation Error: Core  1 panic'ed (IntegerDivideByZero). Exception was unhandled.
Core 1 register dump:
PC      : 0x400d69cf  PS      : 0x00060430  A0      : 0x800e4428  A1      : 0x3ffb1f30  
A2      : 0x00000000  A3      : 0x00000096  A4      : 0x00000000  A5      : 0x000003dc  
A6      : 0x3ffbec04  A7      : 0x3ffc1630  A8      : 0x800d6990  A9      : 0x3ffb1f10  
A10     : 0x00049474  A11     : 0x00000001  A12     : 0x00000000  A13     : 0x3ffc1f00  
A14     : 0x3ffb1ecc  A15     : 0x3ffcf434  SAR     : 0x0000000a  EXCCAUSE: 0x00000006  
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff  

Backtrace: 0x400d69cf:0x3ffb1f30 0x400e4425:0x3ffb1fb0 0x4008932d:0x3ffb1fd0

Rebooting...
```

Seen with current master branch, no GM tube connected, BME280 connected.

ThomasWaldmann, 2019-12-05:
time_difference has still some issues.

Note the integer not float type:
```
unsigned long time_difference;
```
This should be `if/else` and the `if` needs to compare with 0 (not 0.0):
```
Count_Rate = 0;
if (time_difference != 0.0) {
  Count_Rate = (float)GMC_counts*1000.0 / (float)time_difference;
}
```

Here is a not yet fixed divison-by-zero candidate:
```
current_cpm = (int)(GMC_counts_2send*60000 / time_difference);
```

rexfue, 2019-12-08:
OK, will be fixed

rexfue, 2019-12-08:
Why else? Count_Rate is 0.o, that's ok, I guess.

ThomasWaldmann, 2019-12-08:
yes, it is ok, but it should be in the else branch.

rexfue, 2019-12-08:
why? it's less code this way.

ThomasWaldmann, 2019-12-08:
because you are doing an assignment that has no effect in the if-case (only in the else-case)

PR #96 (2019-12-09): check for time_difference == 0 added
fixes #8
closed 2019-12-09 by ThomasWaldmann