ecocurious2/MultiGeiger #11: wrong kit contents or wrong schematic?

ThomasWaldmann, 2019-10-27:
The schematic says:

- C3 100nF
- C4 100pF

My kit contains:

2x WIMA 100/1000 (red 5mm raster foil capacitor, "2,5" printed on the back)

So, what's correct?

rexfue, 2019-10-27:
Also related to wrong (old) PCB version. Sorry for that.

ThomasWaldmann, 2019-10-27:
OK, keep issue open for a while in case someone else also has old PCB.

ThomasWaldmann, 2019-10-27:
Closing, there were only 3 of these boards and their location is known.

ThomasWaldmann, 2019-10-27:
I still think there is an issue.

If you look there, you do not see any 2 capacitors with same value.

So, was my kit contents wrong? Only mine?

rexfue, 2019-10-27:
The Kit-Content is wrong (blame Jürgen :-)  ). Tjalf had the same issue I guess: C2 was missing at his kit.

ThomasWaldmann, 2019-11-09:
OK, guess the inital kit assembling is done by now (and the last kits I received didn't have this issue, so I guess this was a singular mistake). Closing.

closed 2019-11-09 by ThomasWaldmann