ecocurious2/MultiGeiger #12: PCB: adjust pin distance for D4 to 12.5mm

ThomasWaldmann, 2019-10-27:
D4 BYV26E (glass) seems to have already 90deg bent wires fitting into 12.5mm raster.

The PCB v1.1 jb/rxf has 10mm raster, so one needs to re-adjust the diode wires to that raster.
Consider the diode is glass package, one needs to be careful not to damage it.

Would be better if the PCB fitted to default BYV26E raster of 12.5mm.

rexfue, 2019-10-27:
Related to wrong PCB - first test version. Version 1.4 is the actual version.

ThomasWaldmann, 2019-10-27:
Workaround: Can be re-bent using some pliers, be careful, glass!

ThomasWaldmann, 2019-11-23:
Juergen said the diodes usually have straight wires, just these were already bent for some reason.

So this is not a general issue, closing.

closed 2019-11-23 by ThomasWaldmann