ecocurious2/MultiGeiger #14: LND712 GM tube - can it be used for MultiGeiger?

ThomasWaldmann, 2019-10-27:
From another GM kit (offline only, without WiFi, without LoRa) I have a LND712 (alpha, beta, gamma) which could be put online using multigeiger maybe?

Tube specs: https://www.lndinc.com/products/geiger-mueller-tubes/712/

LND712: 500V (recommended. 450V-650V overall), 10M Anode resistor.

Si22G: 400V (recommended. 360-440V overall).

So, just change the Z-diodes to a sum of 500V?

What I wondered: is putting Z-diodes in series and adding the voltage a valid method?

ThomasWaldmann, 2019-10-29:
answering my z-diode in series question:

https://toshiba.semicon-storage.com/us/design-support/faq/diode/is-it-ok-to-connect-multiple-zener-diodes-in-series.html

ThomasWaldmann, 2019-10-29:
Idea:

Currently, 2x 200V Z-Diodes are used to define a fixed 400V "full" voltage.

Alternatively, 4 Z-Diodes with all-different voltages could be used to adapt to different GM tubes by positioning 2 jumpers left or right on 2x 3-Pins.

upper-left: 180V, upper-right: 240V
lower-left: 220V, lower-right: 260V

That would give: 400V, 440V, 460V, 500V.

Just as an example, Z-D voltages should be selected for availability (same Z-diode manufacturer / type, zener current) and GM tube voltage requirements.

ThomasWaldmann, 2019-10-29:
Hmm, anode resistors also need to adapt.

rexfue, 2019-10-29:
OK, but I think, there is no need to change the GM after building the device. And who wants to do it anyway can change the diodes and the resistor. i guess, this will be a seldom case.

rexfue, 2019-10-29:
I asked Jürgen: We think, the LND712 could be used.

ThomasWaldmann, 2019-10-29:
OK, maybe the jumpers are overkill. Just solder in whatever is needed.

BTW, is there a parts list somewhere? In the schematic, there are sometimes specific part numbers, but iirc the Z-Ds were not specified.

I tried searching some Z-Ds on mouser.com, but the huge amount of choices is confusing...

What are the design criteria to select for? Guess <= 500mW, zener current = ?, tolerance = ?, price, through hole, ...?

rexfue, 2019-10-29:
Alles wird gut => Stückliste ist noch veraltet, Jürgen macht ne Neue und stellt sie dann auf GitHub.
Z-Dioden z.B. BZX85C200, die kleine mit 4V7 ist ziemlich beliebig.

rexfue, 2019-10-29:
mistake: no 4v7, use a 3V3 Z_Diode

ThomasWaldmann, 2019-10-29:
Just had a look at reichelt.de's Z-D offers: nothing above 200V.

So, a nice feature of a future pcb could be to offer 3 places for Z-Ds, so 500V can be made e.g. by 200+200+100. Also more freedom for other desired voltages.

rexfue, 2019-11-03:
Yes, sounds like a good idea. There should be space left for one additional diode.

ThomasWaldmann, 2019-11-26:
See also #86 - it seems the LND712 somehow works (although I am not sure about whether the CPMs seen are as they should be - it's hard to compare with the bigger / different tubes and also I do not have a conversion factor yet for the LND712).

ThomasWaldmann, 2020-05-16:
After a long idle time for the LND712, I pulled that out again, flashed current firmware and it works!

It uses the standard PCB and parts, just with 1 addtl. 100V Z diode in series with the 2 on the board, so the tube gets ~500V as required.

It's ticking happily at:
- ~25 CPM (natural background)
- ~500 CPM (with some radiation source)

OK, so I guess the only stuff that is missing is:
- a reasonably calibrated entry in the "tubes" array + #define for the index
- a spec how to send this to sensor.community / madavi so it does not get mixed up with the Si22G

rexfue, 2020-05-16:
sensor.community will not be a problem, we just have to tell Rajko to take in the LND712