ecocurious2/MultiGeiger #18: setup UI: clarify field names

ThomasWaldmann, 2019-10-27:
The field names in the setup menu are all else than clear / self explaining:
```
Current         ---->      Suggested
Thing Name                 Built-in WiFi AccessPoint ESSID (do not change)
AP password                Built-in WiFi AccessPoint Password
WiFi SSID                  WiFi Client ESSID
WiFi passwort              WiFi Client Password
```

If named like that, I guess the docs could be shortened as it is self-explaining.

rexfue, 2019-11-03:
I want to change the GUI anyway. We need to have some settings from userdefines.h to be available in the web interface (i.e. speaker on/off etc).

ThomasWaldmann, 2019-11-14:
As a first step, would the field names as suggested be ok?

ThomasWaldmann, 2019-11-16:
ping?

rexfue, 2019-11-18:
I suggest shorter names
Thing Name                 Geiger-SSID
AP password                Geiger-Password
WiFi SSID                    Home-WLAN SSID
WiFi passwort             Home-WLAN Password

I won't use 'WiFi', because in every (German) mobile phone there it's called WLAN.

ThomasWaldmann, 2019-11-18:
The first 2 are one set of credentials and belong together, so that should be reflected in the field name by using the same wording (except ssid and password, which is the only difference).

Same for the last 2.

Not sure why we need to talk about "thing" (there is TTN for LoRa, but we are talking about wifi stuff here).

ThomasWaldmann, 2019-11-18:
And in general I would just write it in a way so one does not need to read docs to understand it.

ThomasWaldmann, 2019-11-18:
I would not talk about "Home WLAN" or "Home WiFi".

If i install one at my company, it is not my "home".

AFAIK, WLAN is the more popular term in german speaking countries, while WiFi is the usual english term.

rexfue, 2019-11-18:
Explanation:
The names came from the third party library IotWebConf and I wouldn’t change there anything in the first place.

ThomasWaldmann, 2019-11-18:
I assumed the names of the fields are configurable? Didn't look at the code yet.

Also I think we already agreed to use English consistently and not half English, half German?

rexfue, 2019-11-18:
> 
> I assumed the names of the fields are configurable? Didn't look at the code yet.
Don’t know, may be.

> Also I think we already agreed to use English consistently and not half English, half German?
And what is with WLAN not english?

ThomasWaldmann, 2019-11-18:
My mobile phone (en-US) says "Wi-Fi" in the settings (so guess I am right that this is the more usual term in English).

rexfue, 2019-11-18:
But WLAN is an english abbreviation

ThomasWaldmann, 2019-11-18:
yes. "Handy" also sounds english, but only used by german speaking people while english speaking ones say "mobile" or "mobile phone".

ThomasWaldmann, 2019-11-18:
https://docs.microsoft.com/en-us/style-guide/a-z-word-list-term-collections/w/wi-fi

(they do not address Wi-Fi vs. WLAN there)

ThomasWaldmann, 2019-11-18:
Opinion from IRC:
```
11:57  elho$ 
Not a native speaker, but its not wlan (that's more a German thing, like "Handy" for mobile).

Wikipedia uses Wi-Fi and states: The name is sometimes written as WiFi, Wifi, or wifi, but these are not approved by the Wi-Fi Alliance. IEEE is a separate, but related, organization and their website has stated "WiFi is a short name for Wireless Fidelity".
```

rexfue, 2019-11-18:
OK, I don't want to decide it. Lets here the others too.

I guess, our normal user is german (most of them) and - referring to Nicola - he should not be a 'technician'. So I prefer WLAN.

ThomasWaldmann, 2019-11-18:
@rexfue I totally misunderstood your comment where you suggested names due to the missing formatting. I thought you just suggested the new names, but guess you meant `old new`.

I added some `-->` to make it clearer as `old --> new`.

ThomasWaldmann, 2019-11-18:
As a general note, `-` (Bindestrich) is also rather a german thing. In English it is often just a blank.

commit 40520fb8 (2019-11-26): Merge pull request #75 from ThomasWaldmann/iotwebconf-form-labels

clarify field names of web config form, fixes #18
commit 8228fd28 (2019-11-26): clarify field names of web config form, fixes #18
closed 2019-11-26 by rexfue