ecocurious2/MultiGeiger #20: fix Si22G conversion factor

ThomasWaldmann, 2019-10-27:
... after merging #19, see XXX FIXME.

ThomasWaldmann, 2019-10-29:
Related:

https://www.pocketmagic.net/tube-si-22g-c%D0%B8-22%D0%B3-sensitive-geiger-muller/

https://www.uradmonitor.com/topic/hardware-conversion-factor/
https://github.com/cristianst85/rrd-uradmonitor-perl/blob/master/scripts/rrd_uradmonitor.pl

http://arduino-geiger-pcb.blogspot.com/2017/03/conversion-rates-for-soviet-made-geiger.html

http://home.agh.edu.pl/~bartus/index.php?action=efekty&subaction=arduino&item=33

ThomasWaldmann, 2019-10-29:
From current source code:
```
const float GMZ_factor_uSvph_SBM20 = 1/2.47;                //   for SBM-20 (== 0.404858)
const float GMZ_factor_uSvph_SBM19 = 1/9.81888;             //   for SBM19 (== 0.101845)
...
Dose_Rate = Count_Rate * GMZ_factor_uSvph;
```

ThomasWaldmann, 2019-10-29:
based on uradmonitor, i computed: GMZ_factor_uSvph_Si22G = 1/8.59295

ThomasWaldmann, 2019-10-29:
based on arduino-geiger-pcb (cpm -> uSv/h):

SBM20: 1 / 220 == 0.00454545
SBM19: 1 / 928 == 0.00107759
Si22G: 1 / 792 == 0.00126263

ThomasWaldmann, 2019-10-29:
So, 3 geiger counters and 3 quite different sets of conversion factors. Which ones are correct?

rexfue, 2019-10-30:
I think it's always just an estimate. The factor depends on too many influences.
Juergen will tell us the factor for Si22G in the next days.

ThomasWaldmann, 2019-11-03:
There should be also a comment above these lines, where these values come from / how they are calculated.

ThomasWaldmann, 2019-11-03:
@WwwBoehriDe (guess this is you, Juergen?), can you have a look?

WwwBoehriDe, 2019-11-04:
Hi, yes, I am working on that. I can find two values - quite confusing. Bu I think I will come back to this this week.

ThomasWaldmann, 2019-11-12:
Note: because this is yet not fixed in master and we had the Si22G tube, we've seen 5x too high nSv/h values on the correctly counting geiger last Sunday at shackspace.

Lesson: use factor 0.0 next time to be obviously wrong/unsupported rather than displaying confusing/disturbing values.

ThomasWaldmann, 2019-11-12:
Note: the code right now has factor 0.0 for the Si22G (see previous comment).

ThomasWaldmann, 2019-11-12:
Yet another idea! :-)

If we use a `-1.0` conversion factor, the output code could check for negative values and emit `???` or `unknown` instead of the value.

ThomasWaldmann, 2019-11-17:
What we can expect, check this against MultiGeiger displayed values.

https://odlinfo.bfs.de/DE/aktuelles/messstellenliste.html

City and Gamma-Radiation uSv/h 22.11.2019 22:45
Asperg: 0.113
Böblingen: 0.114
Esslingen: 0.103
Sindelfingen: 0.110
Stuttgart: 0.111
Tübingen: 0.097

WwwBoehriDe, 2019-11-17:
I am working on that. There are two differnet values in the internet. The next day I want to make a comparing measurement next to the Sindelfingen ODL-Station (I hope I can find it  :-(  ).

ThomasWaldmann, 2019-12-05:
@WwwBoehriDe did you already try finding it? the map on the bfs site seems to be rather precise.

WwwBoehriDe, 2019-12-06:
Hi, yes, I did measure and compare it directly at at the governments ODL-Station Sindelfingen. So I was able to calculate the correct value for the Si22g. There was no good value in a datasheet  :-(.
The Constant for the Si22g geiger tube I measured is:
44205 Counts per uSv...

So the Resulting factor for the sofware is:
0.08143875127248        (uSv/h) per (Counts per Second)

ThomasWaldmann, 2019-12-07:
Ah, great. And it's even in between the other 2 known values. :-)

I'll update the code...

ThomasWaldmann, 2019-12-07:
See #94.

commit 2436ed03 (2019-12-09): Merge pull request #94 from ThomasWaldmann/fix-si22g-factor

correct Si22G conversion factor, fixes #20 fixes #94
commit 5da10457 (2019-12-09): correct Si22G conversion factor, fixes #20

thanks for Juergen for determining the factor!
closed 2019-12-09 by rexfue