ecocurious2/MultiGeiger #23: voltages seen on GMZ_count connection (PCB v1.4)?

ThomasWaldmann, 2019-10-29:
While reading the schematic I was wondering about voltages seen on that connection?

It comes from C4 coupling it to the high voltage part and leads to a ESP32 input pin.

Considering R2 : R3+R4 ratio and C3 voltage of about 400V, the pulse could be 400V -> 340V (left hand side of C4) and the AC part of that about -60V (right-hand side of C4 in the schematic), right? This assumes that the GM tube has about no resistance in the moment of a pulse (or low resistance relatively to the rather high other resistor values).

Is there some internal protection in the ESP32 capable of enduring this?
What about if JP1 is used to connect to some other uC?

Or do we need 2 more clamping diodes to make sure this is between GND and Vcc levels?

rexfue, 2019-10-29:
We rely on the internal clamping diodes.

ThomasWaldmann, 2019-10-31:
Suggestion: put clamping diodes onto the GM part of the PCB (to the right of the cutting line in case the ESP32 part gets cut away).

ThomasWaldmann, 2019-11-23:
Juergen said that usually today's microcontrollers have ESD protection diodes, so this should work without diodes on our part.

Just do not connect the HV part directly to anything without own protection diodes.

closed 2019-11-23 by ThomasWaldmann