ecocurious2/MultiGeiger #24: PCB: make cut between uC and GM high voltage part possible

ThomasWaldmann, 2019-10-29:
I mean a straight vertical cut just to left-hand side of the JP1 5pin GM-part-of-the-pcb connector.

https://github.com/ecocurious/MultiGeiger/blob/master/docs/Board_V1.4.pdf

The PCB v1.4 is already made in a way so this is **almost** possible.

Just C2 is a bit in the way, so a bit more space for cutting in between the ESP32 part and the GM part would be good.

That way, we could reuse the GM part of the PCB when wanting to use a different uC (now or one day when the esp32 is considered obsolete).

Would also open the doors for people prefering other uCs right now or having already working measurement systems and just not wanting to design their own PCB for the GM part.

rexfue, 2019-10-29:
I agree. You can cut left of C2 using the two mounting holes. 
I cannot shift all pieces to the right because the max width of the board can only be 100mm!

ThomasWaldmann, 2019-10-29:
Ah, right, didn't think about the 100mm standard size.

OK, guess cutting there works. Thanks!

closed 2019-10-29 by ThomasWaldmann