ecocurious2/MultiGeiger #27: remove pointless comment

ThomasWaldmann, 2019-11-03:
In the source, there are quite some not really needed comments (while needed comments are missing at other places).

If the comment text just repeats what's already clearly seen in the code, I would suggest removing it as it is just a burden and not useful.

Examples:
```
isr_GMZ_counts++;  // Count

GMZ_counts = isr_GMZ_counts;  // copy values from ISR 

u8x8.setFont(u8x8_font_5x8_f);     // small size

clearDisplayLine(7);                    // clear line

u8x8.drawString(0, 7, txt.c_str());     // print it

cycle_time_us   = 1000000000/frequency_mHz;       // calculate all we need

int httpResponseCode = http.POST(body);                   //Send the actual POST request
String response = http.getString();                       //Get the response to the request
Serial.println(httpResponseCode);   //Print return code
Serial.println(response);           //Print request answer

...
```

rexfue, 2019-11-09:
It's more burden to delete them than let them in! They don't bother me.

ThomasWaldmann, 2019-11-09:
I can delete them if we agree they are pointless and could be deleted.

rexfue, 2019-11-09:
ok

ThomasWaldmann, 2019-11-09:
BTW, I will be in shackspace tomorrow afternoon (~ after 14:00) and try to get the geigers assembled and running. Maybe some other guy from Neckarwestheim will join me and assemble one of them.

rexfue, 2019-11-09:
Viel Spaß und viel Erfolg!

ThomasWaldmann, 2019-11-14:
Doing this now.

commit 4e609d37 (2019-11-14): Merge pull request #55 from ThomasWaldmann/remove-pointless-comments

remove pointless comments, fixes #27
commit a1003563 (2019-11-14): remove pointless comments, fixes #27
closed 2019-11-14 by ThomasWaldmann