ecocurious2/MultiGeiger #29: global variables with 1 char name: t, h, p

ThomasWaldmann, 2019-11-03:
Not good style and can't be searched for easily.

ThomasWaldmann, 2019-11-03:
```
      t = bme.readTemperature();
      h = bme.readHumidity();
      p = bme.readPressure();
```

So, rename them to temperature, humidity, pressure?

ThomasWaldmann, 2019-11-07:
there is change relating to this in PR #30.

rexfue, 2019-11-09:
shall I close this (and other solved) issue(s)  or will you do it?

ThomasWaldmann, 2019-11-09:
I can do it after it is in master.

I a commit specifically fixes some github issue, one can also write a commit comment like:
```
some comment that makes sense, fixes #123
```
Then github will auto-close the issue when the PR is merged into master.

rexfue, 2019-11-09:
OK understood

ThomasWaldmann, 2019-11-12:
was merged, so this is fixed.

closed 2019-11-12 by ThomasWaldmann