ecocurious2/MultiGeiger #31: git: tags for releases are missing

ThomasWaldmann, 2019-11-07:
On githib, if you click on "Code" -> "Branch" -> "Tags", you see there is only one tag define yet.

Guess there should be a tag for everything that is mentioned as a separate version in the change log.

ThomasWaldmann, 2019-11-07:
btw, to not forget such stuff, it is a good practice to have some "how to make a release" in the developer docs of a project.

this also would help making a release in case the person normally doing it has no time to do it.

rexfue, 2019-11-09:
OK, tags are missing. In future we will add tags.

ThomasWaldmann, 2019-11-12:
1.9 tag also missing.

rexfue, 2019-11-12:
Shit, i knew, i will forget something.

ThomasWaldmann, 2019-11-12:
see #46.

not sure if it is important to have tags for past / current releases?

if so, keep this open. otherwise, we can close this ticket.

closed 2019-11-26 by ThomasWaldmann