ecocurious2/MultiGeiger #32: use other pins to not need external pullup resistors?

ThomasWaldmann, 2019-11-08:
From the ESP32 docs:
```
GPIO34-39 can only be set as input mode and do not have software pullup or pulldown functions.
```

That's were PCB 1.4 has the dip switches and the external 10k pullup resistors.

If we could use some other pins, we could use internal (sw configurable) pullups and need to buy / solder / have PCB space for 4 resistors less.

ThomasWaldmann, 2019-11-08:
https://esp32.com/viewtopic.php?t=5111

Internal: about 50kOhm (between 10k and 100k).

ThomasWaldmann, 2019-11-08:
When considering what the lora heltec device uses for other purposes, these GPIOs seem still completely unused:

1 io
3 io
13 io
17 io
32 i
33 i

rexfue, 2019-11-09:
OK, good idea. But wrong!

We want to use 3 Chips: Heltec Wifi Kit 32, Heltec Wireless Stick and Heltec Wireless Stick lite.
And among these 3 there is NO Pin free (see attached Excel sheet).

rexfue, 2019-11-09:
Ups sorry, missed the Excel file.

ThomasWaldmann, 2019-11-09:
OK, I only looked at the Heltec Wifi and Wifi/Lora uC boards.

closed 2019-11-09 by ThomasWaldmann

ThomasWaldmann, 2019-11-09:
Still would be interested in the spreadsheet. :-)

rexfue, 2019-11-09:
BTW: the switch is only optional- That means, the average user has no need to solder / buy  the the switch and the resitsiors.

rexfue, 2019-11-09:
should be on the way (sent 22:33)

ThomasWaldmann, 2019-12-17:
Link:

https://github.com/ecocurious/MultiGeiger/blob/master/docs/hardware/Pinning_of_HeltecBoards.pdf