ecocurious2/MultiGeiger #34: original files for pcb and schematic missing

ThomasWaldmann, 2019-11-09:
in the docs directory, there is:

-  the pcb (as pdf, not as original file)
-  the schematic (as pdf, not as original file)

Would be great to also have the original files (editable), not only the pdf (read-only).

rexfue, 2019-11-11:
Do you think that makes really sense? They are EAGLE-Files, and not all people use EAGLE.

ThomasWaldmann, 2019-11-11:
Well, while I personally would have preferred it to be in some open format editable with FOSS software, Eagle files are better than nothing.

And it is not the point that not everybody uses Eagle, but whether one has access to editable files which are available under some FOSS license so one can produce original or changed PCBs (after installing Eagle oneself or finding some other Eagle user on this planet).

rexfue, 2019-11-11:
I don’t know what FOSS is. Can you please explain?

ThomasWaldmann, 2019-11-11:
Free (and) Open Source Software.

ThomasWaldmann, 2019-11-11:
don't have practical experience with FOSS PCB SW, but maybe someone else can comment on this.

ThomasWaldmann, 2019-11-12:
fixed by https://github.com/ecocurious/MultiGeiger/commit/fbffb56ccbf69f0ee8f3396aefb131396e4bcd5d

closed 2019-11-12 by ThomasWaldmann