ecocurious2/MultiGeiger #35: use pwm for sound?

ThomasWaldmann, 2019-11-09:
There is some code toggling GPIOs by software to make the tick sound and some startup sound.

The ESP32 has PWM hardware that can be programmed with frequency, ratio and then runs completely in hardware without using the CPU for toggling pins or doing delays.

Currently, the piezo is connected to 2 GPIOs that are driven inverted to each other to cleverly double the amplitude (by going from hi/lo to lo/hi). Is that needed to be loud enough or would half of that also suffice?

Not sure if we could easily have 2 PWMs that are offset so that they come out inverted.

Note: we could use single PWM even with the current PCB, wasting 1 IO pin by just pulling it low permanently while the other is doing PWM.

rexfue, 2019-11-11:
Good idea.
I will test it, but think it doesn't have any advantage (concerning code size).
At the moment, we don't want to change the software (beside fixing bugs), because want to hold a workshop at the beginning of December.

ThomasWaldmann, 2019-11-11:
OK, fixing just issues for now is fine for me.

So let's delay this until later.

WwwBoehriDe, 2019-12-06:
I think if you only use one pin the Audio-Power would be 1/4 and it would be not loud enough. And acctually I was pretty amazed how good and precise my simple frequency genartor loop worked. The ESP32 is, regarding speed, really a rocket  :-)

ThomasWaldmann, 2020-05-03:
Note: when trying to slow down the fast-spinning main loop (about once per 1-2ms), i noticed that it currently can not be really slow because the tick/blink is generated there. the slower it spins, the less random sounding the ticks get (and also it does not tick once per GM event any more).

If the ticking would not depend on mainloop spinning, but would somehow "run in the background", the main loop could spin way slower (like once per second or so).

If the tick sound would be generated by PWM, maybe the ISR could enable the sound and set a timer which disables the sound a bit later again.

WwwBoehriDe, 2020-05-03:
I think using the PWM-Generator is not possible because this is not able to drive two pins complementary (as far as i know). But two pins complementary pins are required to have enough speaker volume. 

I am not sure how this module could run in background. But probalby that is a good idea if it is possible. But then the Interprocess-Communication should not be cryptic but it should be obvious even to programming starers  :-)

ThomasWaldmann, 2020-05-04:
Some links:
- https://esp32.com/viewtopic.php?t=12654
- https://forum.micropython.org/viewtopic.php?t=3717&start=10
- https://forum.micropython.org/viewtopic.php?f=2&t=7774&start=10

docs:
- https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/mcpwm.html
- https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/ledc.html

I never used mcpwm yet (motor control pwm), but the external motor driver circuit schematic looks like A/B outputs need to be inverted, right? So, just what we need for our speaker?

ThomasWaldmann, 2020-05-04:
https://www.espressif.com/sites/default/files/documentation/esp32_technical_reference_manual_en.pdf

figure 110 on page 426 is about what we need (besides that offset, which is likely an advanced feature we just don't need).

rexfue, 2020-05-04:
It looks like it meets our requirements.
But is it worth the effort? Because of the 3sec sound at the beginning? I would rather delete the sound than write new software.

ThomasWaldmann, 2020-05-04:
I've implemented working PWM sound, see comments in PR #251 about what could be done with it in future.

t-pi, 2020-05-04:
Another suggestion: 
The loop timer could be set as a function of speakerTick. If ticks are not activated, the loop could run at the much lower period mentioned above.

ThomasWaldmann, 2020-05-05:
@t-pi yeah, if neither speakerTick nor ledTick are active the main loop could spin slower.

commit 143eaf98 (2020-05-12): timer driven audio/led sequencer, PWM sound, fixes #35

hardware PWM (with complementary outputs) for sound (ticks and tones).

timer driven audio sequencer, supports tones, speaker and LED "ticks".

can be fed with a list of (frequency, volume, led, duration) integers.

geiger mueller "tick" at 5000Hz, debug "tock" at 1000Hz.
commit fdee72f2 (2020-05-12): Merge pull request #251 from ThomasWaldmann/pwm

timer-driven sequencer, hw PWM sound, fixes #35
closed 2020-05-12 by ThomasWaldmann