ecocurious2/MultiGeiger #36: LICENSE missing

ThomasWaldmann, 2019-11-09:
This repo has no documented license.

According to german (european?) law, this makes it by default a proprietary non-free project (which is an unfortunate state, because all contributors who prefer FOSS would need to stop contributing).

I guess this is not intended, but the license was just forgotten...

Some of the usual licenses:
- BSD license
- MIT license
- GPL license

MIT and BSD are rather short and simple (almost "do what you want").

GPL is longer and more restrictive in the sense that it requires the software to stay free (which is a good thing).

The usual way to put up a license is in a toplevel LICENSE file, as seen e.g. there:

https://github.com/borgbackup/borg/blob/master/LICENSE

rexfue, 2019-11-11:
OK, GPL will be added

ThomasWaldmann, 2019-11-11:
There needs to be that LICENSE file and a corresponding (c) remark (at least) in multigeiger.ino, mentioning the license name pointing to that file.

I can make a PR if you like...

rexfue, 2019-11-11:
Not necessary, thank you. I copied the part from Feinstaub.

ThomasWaldmann, 2019-11-11:
Maybe we should get your last PR finished / merged first or it might conflict again with these changes...

rexfue, 2019-11-11:
no problem at the moment - i did my changes in in a local branch for now

ThomasWaldmann, 2019-11-11:
It will still conflict, no matter whether you have published it or not.

And always publish / merge quickly.

If one keeps stuff local on ones disk for longer time (or if there are unmerged PRs) and there are multiple people frequently working at stuff, one will cause more work for fixing conflicts (either for oneself or for other devs).

rexfue, 2019-11-11:
I will try to merge tomorrow evening or Wednesday

ThomasWaldmann, 2019-11-12:
LICENSE (GPL) was recently added.

One bit is missing still, there needs to be a copyright notice also, I'll add that.

ThomasWaldmann, 2019-11-12:
The last bit was fixed by #43.

closed 2019-11-12 by ThomasWaldmann