ecocurious2/MultiGeiger #38: measurement interval / display update interval too short

ThomasWaldmann, 2019-11-12:
Last sunday, I assembled 3 geigers with friends at shackspace.de.

When seeing them work, I noticed that there is a problem with how measurements are done / displayed. It seems that the code is rather keen on quickly displaying ....... something.

So the first value after powering up appeared only a few (N) seconds later. As it displays CPM and does not even measure a full minute for that, it has to compute cpm = counts * 60 / N.

This is bad as due to the random of radioactive decay (and maybe also other reasons in physics) and the big variation in short measurement periods for cases of low (normal) radiation, the resulting cpm output varies hugely and thus is pretty useless.

Assuming N == 6 and a real CPM value of 20 (like SBM-20 and background radiation):

```
time [s]  counts     displayed CPM
----------------------------------------------
0..6      5                       50
6..12     1                       10
12..18    2                       20
18..24    0                       0
24..30    2                       20
...
```

So the only thing more or less known after these displayed values is "ok, it's not zero and it is also not 1000 CPM, but maybe something around 20", but you can only tell that by smoothening the values in your mind.

A better and still simple way would be to measure longer, update less frequently.

Or even better, display an average value over the last minute (sliding windows over measurement series) and do not display anything before "enough" pulses have been counted or one minute is reached.

AND maybe do not

ThomasWaldmann, 2019-11-12:
Or: adjust sliding window width over measurements to total counts seen in that window.

E.g. if we only get 20 CPM, let the window be 1 minute wide to have some precision.

If we get 200 CPM, the window could be 6s to get similar precision.

If we get 2000 CPM, we could have 0.6s window for similar precision.

If we get 20000 CPM, we could still have 0.6s window as we do not need to update the display any quicker than that.

rexfue, 2019-11-13:
At the moment we have 10sec Display and calculating interval. I think, that's OK.
If you switch it on for the first time and you have to wait 60sec until you see something - that is frustrating.

ThomasWaldmann, 2019-11-14:
I disagree about the "OK". But I guess it depends whether you see it as a measurement device or something for show purposes. I don't expect high precision, but I also don't expect display of values that aren't worth much. :-)

10 secs, Si22G (relatively big tube), normal german background radiation means 7 or so pulses. With a SBM20 (or other smaller tube), it would be only 2 pulses.

That can easily vary widely, just by some random pulses more or less, so what one gets is a display value that jumps arounds significantly.

So while I agree somewhat to the frustrating aspect of having to wait for a longer time, I still think it is worthwhile doing so. Except if there is high radiation and a good value can be displayed quicker without losing precision.

ThomasWaldmann, 2019-11-14:
BTW, there is one more aspect to it:
- display: one has to wait for the display to update, but (my) expectation is that displayed values are somehow precise (or at least not jumping around wildly)
- sound: here the user gets immediate feedback. if it ticks a little, you know that there is little / normal radiation - even without having something on the display. if it ticks a lot, you know there is more.

WwwBoehriDe, 2019-11-16:
Hi, I know the point. The Display is always a compromise between 
- precission and
- speed
So made the CPM-Display fast reactive. But for the case you want to measure precise all counts and measurement time are summed up and calculated to uSv/h. 

I think for the 10 seconds window its important to measure the time between pulses - but additionally have a timeount.

I think that making both values precise but slow is no good idea.

ThomasWaldmann, 2019-11-16:
Well, in my case the nSv/h value was way off, see https://paste.thinkmo.de/bwmaX6J9 (i am only considering the multigeiger in the middle, the others are not counting correctly yet). For germany, this should be rather 0.2 uSv/h, but it displays 1.1 uSv/h.

This is why I am caring for a usable CPM rate also (so one can compare it to what the manufacturer gives in their datasheet).

I have nothing against quick display updates IF there is sufficient data (in our case: if there are enough pulses). But if there isn't, it is useless to have a widely varying display (and having to compute the medium value of CPM in your mind instead of on the uC).

As the software also supports smaller tubes like the SBM20 (which only have 20CPM under normal conditions), the problem would be even more serious there.

ThomasWaldmann, 2019-11-16:
(the software used for the devices on the photo was master branch of 1w ago)

ThomasWaldmann, 2019-11-17:
I'll give a specific example what the current master branch shows to me (in order):
85 73 51 60 75 69 96 91 102 67 109 76 66 91 66 86 49 113 54 73 60 89 ...

So, displayed CPM is varying by a factor of over 2 (49 -> 113).

ThomasWaldmann, 2019-11-17:
With DISPLAYREFRESH == 30000 [ms]:
73 66 76 62 89 107 74 81 71 98 89 86 86 91 75 84 73 88 ...

With DISPLAYREFRESH == 60000 [ms]:
63 106 82 92 81 88 78 87 96 75 ...

WwwBoehriDe, 2019-11-17:
... So, displayed CPM is varying by a factor of over 2 (49 -> 113).

Hello Thomas,
yes, thats exactly what I would expect. Normally (95 % of the values) should be in twice the standard deviation. So it comes out out to: 77 +/-43 
=> 34 ... 120 CPM

Like I said: Thats the tradeof between speed and accuracy which is nealy always critical with geiger counters. Thats the reason why I have chosen such a big geiger counter tube: to get as many pulses as possible. And now we get about 13 pulses per counting interval.

But I would like to go for the 10 Seconds. An other Idea is to make it software adjustable (gliding average). But I think for the first time I would leave it like it is. Otherwise its too confusing.

ThomasWaldmann, 2019-12-09:
making this more flexible might depend on #66.
removing it from milestone due to that.

ThomasWaldmann, 2020-05-09:
after my recent refactorings in #247, this is now trivial to change via DISPLAYREFRESH in multigeiger.ino.

note that the "Serial_Logging" output updates at the same time as the display is refreshed.

closed 2020-05-09 by ThomasWaldmann