ecocurious2/MultiGeiger #39: random reboots?

ThomasWaldmann, 2019-11-12:
I've seen the geigers playing the startup sound multiple times after plugging into USB.

Random reboots? Fatal exceptions leading to reboots? Power? HV issues?

rexfue, 2019-11-13:
Don't know what causes that, haven't seen it yet.

ThomasWaldmann, 2019-11-16:
Have it seen today again, had the MultiGeiger running for some hours and it rebooted multiple times (noticeable due to the startup sound).

Was indoors at room temperature, located without case on a table (plastic surface).

ThomasWaldmann, 2019-11-23:
Note: always have a look at serial/usb output when such stuff happens.

Also had reboots today, see #82 - as @rexfue is already working on the ISR stuff, I am closing this for now.

closed 2019-11-23 by ThomasWaldmann