ecocurious2/MultiGeiger #40: multigeiger soldering for dummies :-)

ThomasWaldmann, 2019-11-12:
# Polarity / orientation matters
- diode polarity matters! ring side == where the arrow of the diode symbol points to. black cases have white ring, transparent or white case have black ring.
- 100uF (black cylinder) is a electrolytic capacitor, polarity matters (long wire == plus, also see `-` mark on minus side)
- the fuse holder for the plus pole of the tube is asymetric! check fit.
- the heltec wifi kit 32 usb connector is above the piezo.

# Polarity does not matter
- the smaller (red block) capacitors are foil type
- coil (black, ferrite core)
- resistors
- piezo

# truncating the esp32 connectors (female, on geiger pcb)
- for heltec wiki kit 32, one needs to get rid of 2 connector pins
- to do that use a wire cutter and cut through the middle of the 2nd connector pin
- that way the 3rd one will still be healthy (maybe not pretty)
- point the not-pretty side to the end of the geiger pcb, not to the big piezo (might not fit)

rexfue, 2019-11-13:
Jürgen writes (has almost written) detailed instruction to build the hardware.

ThomasWaldmann, 2019-11-14:
I don't see anything?

If one wants to work collaboratively without wasting resources to doing the same work twice, one should not work "in the dark".

ThomasWaldmann, 2019-12-21:
https://github.com/ecocurious/MultiGeiger/tree/master/docs/Bauanleitung

TODO: compare the stuff above against the linked docs, update the docs as needed, close this issue.

M-a-r-i-a-n-n-e, 2021-03-20:
@ThomasWaldmann Now the Bauanleitung is also on ReadTheDocs, maybe you want to add your text there?