ecocurious2/MultiGeiger #46: how to release

ThomasWaldmann, 2019-11-12:
- check if everything is nice and working
- check if there are trivial things still open in the issue tracker, consider doing it.
- check if there is something left in the milestone for the next release, do it or move it to later.
- update the version number(s) in the code for the next release
- add a changelog entry with all stuff that's new in that release
- tag the release with git (and later make sure the tag is actually pushed to github)
- optional, but nice: do a release on github (see "releases" tab)

ThomasWaldmann, 2019-11-12:
anything missing?

rexfue, 2019-11-13:
I don't think so.

ThomasWaldmann, 2019-11-14:
OK, so the above text can be used to make a PR for docs/development.txt (or a wiki page).

commit cfdb7bfb (2020-02-09): updated/fixed development docs, fixes #46
closed 2020-02-09 by ThomasWaldmann