ecocurious2/MultiGeiger #47: milestones

ThomasWaldmann, 2019-11-12:
there is a nice feature on github, called milestones.

they can be used for release planning ("what shall get into next release?") by assigning tickets to milestones.

if all tickets in a milestone are closed (or it has been decided to remove them from that milestone), the code is release-ready.

so, what's the next release? 1.10?

rexfue, 2019-11-13:
I think, all these pretty nice tools and possibilities are a little overkill for our little geiger counter.

Yes, I think we call it 1.10.

ThomasWaldmann, 2019-11-14:
The point about milestone is to make goals documented.
If multiple people work on a project, it is good if everybody knows the goals for release x.
So they know what to do / what not to do / what to do later and can help.

Also, one will notice when the goal is reached and stuff is release-ready.

ThomasWaldmann, 2019-11-14:
As we might have an assembly at 36c3, I added a milestone for stuff that should be done until then.

ThomasWaldmann, 2019-11-23:
milestones are also covered in #56, so this issue can be closed.

closed 2019-11-23 by ThomasWaldmann