ecocurious2/MultiGeiger #48: setup travis-ci

ThomasWaldmann, 2019-11-12:
of course we can't test the software with the hardware there, but we could:
- check if it compiles(?)
- do some style checks on the source code (e.g.: no tabs, no trailing blanks)
- maybe have some tests that can work without the hardware / with mocked hardware

i use travis-ci for python projects. it's quite nice to have a CI system that does stuff automatically for each PR. but i have 0 experience with travis-ci for C / ESP32 / hardware projects.

rexfue, 2019-11-13:
Never heard about this tool.
I think, this project is small enough to build the hardware and test with it.

ThomasWaldmann, 2019-11-14:
travis-ci is quite popular (although far from perfect) and github has nice integration with it.

so all that is automated there will just get run automagically as soon as a PR is made or merged.

ThomasWaldmann, 2020-02-10:
what also could be done on travis-ci is automated style checks (e.g. via `astyle`, see #183).

ThomasWaldmann, 2020-02-11:
done, see #188.

closed 2020-02-11 by ThomasWaldmann