ecocurious2/MultiGeiger #52: write real docs?

ThomasWaldmann, 2019-11-13:
A good README is better than nothing, but for the github page we rather need something else:

Kind of an "elevator speech" - why you want to use THIS Geiger Counter and not another one. Why you want to be part of this community. [...] Basically, the stuff needed before the decision "yes, I want to do that" or "no, this is not what I searched for". And, important: not much else!

Especially, there should **not** be a description of how to install, configure, ... the stuff. Nobody reads long docs when just trying to decide. That stuff should be in our well-structured docs, which we do not have yet.

We could do 2 things for the docs:
- use the github wiki (pro: easy, con: can get messy)
- use sphinx and write docs in reST format and automatically publish them on readthedocs.io for each of our releases (this is more work to set up, but has the best result, if done right).

This is an example of how it can look like when made with sphinx / rtd: https://borgbackup.readthedocs.io/

rexfue, 2019-11-13:
There are many good software repos out that have installation instructions in the ReadMe (for example docker nginx by jwilder). 
But  I think, it would be OK to move these instruction to the wiki. To use (and lern) a new tool is imho too much effort; therefore i disagree for that.

ThomasWaldmann, 2019-11-14:
There is a lot of good software with bad (or sub optimal) docs, yes. :-)

ThomasWaldmann, 2019-12-21:
There is some stuff here now (in german):

https://github.com/ecocurious/MultiGeiger/tree/master/docs/Bauanleitung

Some more needs to be read in the README (also german).

Problem: the first one (office document) is not good to handle via git as it is a binary format.

ThomasWaldmann, 2019-12-22:
Ånother problem: it needs to point to the github README multiple times, as a random part of the docs is there.

ThomasWaldmann, 2020-01-02:
Another problem with the current docs: no easy "diff" display with office formats, plain text with markup (see above) would be better.

ThomasWaldmann, 2020-01-16:
meeting: do it.

ThomasWaldmann, 2020-02-11:
some stuff is now on https://multigeiger.readthedocs.io/ (source format: reST markup, see docs/source/ folder, in English).

but it is far from complete and some stuff only exists in german and only in non-text, binary formats.

ThomasWaldmann, 2020-05-22:
https://multigeiger.readthedocs.io/ now has both german and english docs, one can switch between releases and languages. infrastructure is in place enabling us to even add more languages (which i think we should **not** do before having a bigger community for addtl. languages).

the "elevator speech" kind of README is available in english and also translated to german.

i think we should proceed now with:
- we need to move the huge all-in-one german readme to a different place, it should not show up on the github project main page
- we should create a new **very** simple github README to show up on the github project main page
- this should have links to the english and german readthedocs-based documentation
- as long as most stuff is still in the german all-in-one README, it should also link to its new position
- as long as a lot of stuff is still in libreoffice documents, it should also link to them

PR #328 (2020-05-23): new github "front page" README
- point to the translated docs on RTD (en + de)
- point to legacy "README" (de)
- point to legacy odt files (de)

fixes #52.
closed 2020-05-23 by ThomasWaldmann