ecocurious2/MultiGeiger #54: git(hub) crash course?

ThomasWaldmann, 2019-11-14:
Would a git(hub) crash course be interesting for the other developers / contributors?

@WwwBoehriDe @rexfue @ecocurious

I use git (and other DVCSes) and github since quite a while, so I could offer an introduction (preferably at shackspace).

ecocurious, 2019-11-14:
für mich schon interessant.

WwwBoehriDe, 2019-11-14:
Ja, wäre für mich auch interessant.

Gruesse
   Juergen

ThomasWaldmann, 2019-11-15:
https://dudle.inf.tu-dresden.de/bBi2k4b3Nw/

hab mal ein paar termine sa/so rausgesucht fuer den shackspace.

am 24.11.2019 ab 14:30 waere dort eh weiterbasteln / kalibrieren / usw. angesagt an den 3 geigern, die wir dort vor ner woche zusammen gebaut haben. von daher hab ich an dem tag mal die zeit davor in das dudle reingepackt fuer den github crash kurs. wer will, kann dann auch laenger bleiben...

ThomasWaldmann, 2019-11-17:
Fehlt nur noch @ecocurious (Nicola). Kannst Du zu einem der sich dort abzeichnenden Termine?

Wenn nicht, koennen wir uns auch gerne mal separat davon im shack zusammen setzen, wenn Du moechtest.

ecocurious, 2019-11-18:
geht auch teamviewer?

ThomasWaldmann, 2019-11-18:
@ecocurious koennte man ausprobieren (ich hab allerdings tv nur unter windows und development kram mach ich unter linux...).

ThomasWaldmann, 2019-11-19:
Also aktuell sieht es im dudle nach dem Sa 23.11 oder So 1.12. jeweils ab 15:00 aus, das hat bisher die meiste Zustimmung gefunden.

Wobei Nicola im dudle bisher noch fehlt...

ecocurious, 2019-11-19:
worum gehts denn?

ThomasWaldmann, 2019-11-19:
Einfach das Ticket auf Github lesen, dann wird es klarer als in der Notification-E-Mail.

Link ist in der Mail.

ecocurious, 2019-11-19:
ich arbeite am kommenden Sonntag in der Nähe von Tripsdrill und könnte gegen 12 Uhr nach der Arbeit bei dir im Büro / BiBi vorbeikommen.
vielleicht kann jürgen auch dorthin kommen? Ich könnte ihn auf dem rückweg mitnehmen. 

ob ich am 1.12. arbeiten muß und ewo, steht noch nicht fest. 

ansonsten teamviewer

ThomasWaldmann, 2019-11-19:
So 24.11. um 12 bin ich grad beim Mittagessen und muss kurz danach auch schon los Richtung shackspace, weil ich da nen Termin habe.

ecocurious, 2019-11-19:
dann schlage ich teamviewer vor

ThomasWaldmann, 2019-11-21:
Samstag shackspace
==================

Termin zum "Github Crash-Kurs":

Samstag 23.11. ab 15:00 im shackspace (S-Wangen, Ulmer Str. 255),
Seminarraum (wenn frei, anonsten trifft man sich in der Lounge).

Laut dudle sind wir dann zu dritt im shack.

Ich komme mit dem Velomobil und versuche puenktlich zu sein.
Falls ich mich je verspaete bitte ggf. etwas warten.

Vorbereitung:

Bitte git schonmal installiert haben (wenn moeglich) und
wenn ihr ne GUI braucht (nicht mit Kommandozeile arbeiten wollt), die
bitte auch schon vorher installieren. Wenn man "git" auf der
Kommandzeile eintippt, sollte die Hilfe kommen.

Ich persoenlich arbeite mit Linux und git auf der Kommandozeile - die
Kommandozeilenbefehle sind aber bei Windows git genauso.

Die GUIs bilden im Endeffekt auch nur die Kommandozeilenbefehle ab.

WwwBoehriDe, 2019-11-21:
alles klar. Freue mich drauf. Bis dann.
Viele Gruesse
    Juergen

-- 
Viele Gruesse
      Juergen Boehringer

=====================================================================
\        Juergen Boehringer                                         /
/        Austr. 8                                                   \
\        71034 Boeblingen                                           /
/        Tel.:     +49 7031 286207                                  \
\        Mail:     juergen@boehri.de                                /
/        Homepage: www.boehri.de                                    \
=====================================================================

ThomasWaldmann, 2019-11-23:
Done, thus closing for now.

We can repeat this at a later time, for new project members or those who could not come.

I used a whiteboard to explain git, can't be easily done via teamviewer I guess.

closed 2019-11-23 by ThomasWaldmann