ecocurious2/MultiGeiger #56: improve online collaboration

ThomasWaldmann, 2019-11-14:
Some hints about how we could improve collaboration, speed up development, get the burden distributed on more shoulders:

- rely less on synchronuous offline communication / work (like meetings, phone calls), it is slow and does not scale, do more online (as far as possible). you can still meet/call, but put results online.

- be open / transparent, publish early, pull / push often (code, docs)

If stuff only exists in your mind or on your local storage, noone except you can see it, can help with it and, worse, if people start working on the same stuff, double work will be done.

Don't keep commits local for a longer time - at least push them to an open pull request, so you're not working in the dark.

Rather work together on github issues than via e-mail (except if it is private).

- put TODOs / ideas into github issues

Having an issue per TODO or idea (and not just for a bug) is good because then you can get feedback on it, suggestions or someone who has time/interest might just do it.

Also, by writing "working on it!" into that issue, it will be clear that someone works on it (project members can also use "assign to myself"), double work is avoided, progress is more clear.

- create milestones from issues

Milestones can be used to set goals (for a specific hw/sw release, or for sw/hw related preparations for some event) and it will be always clear what is left to do and how far it has progressed yet.

- avoid merge conflicts

This is a big problem here as the project has only a few files where all the work is done (README.md, multigeiger.ino). So, if 2 people work in parallel and do bigger changes, chances are quite high they will create a merge conflict (by editing the same lines).

This can be avoided by:
- always first pull from master before doing own changes in a feature branch
- doing small, focussed changes, publishing quickly (to a PR), wait for review, merge quickly to master
- if there are open PRs (especially if they are easy/trivial), review and merge quickly.
- it is good practice that developer X makes the PR, developer Y reviews and merges it.
- OTOH, developers do not want to wait too long for review (especially if it is blocking them due to merge conflicts), so they might also merge on their own.

ecocurious, 2020-01-02:
ja, wir sollten das bei unserem nächsten meeting festlegen

ThomasWaldmann, 2020-01-02:
@ecocurious dir ist schon klar, dass bei einem offline-meeting v.a. die offline-leute dann da sind und das fuer eine festlegung dann schon nen starken bias bedeutet?

von daher wuerde ich vorschlagen, ihr benutzt fuer abstimmungen ein medium, dass auch online benutzbar ist, z.B. dudle (kann man von github aus verlinken). koennt ihr dann auch von dem treffen aus benutzen.

ThomasWaldmann, 2020-05-22:
Guess everybody interested got this information, so nothing left to do here.

The merge conflict situation also got a bit better since the code is more modularized.

The docs are still a big todo, but over time, we could also get more modularized plain-text docs and thus can also deal more easily with them.

closed 2020-05-22 by ThomasWaldmann