ecocurious2/MultiGeiger #57: TUBE_UNSUPPORTED == 0

ThomasWaldmann, 2019-11-14:
The software uses 1,2,3 for SBM20/SBM19/SI22G.

There is also a "Radiation unknown" entry at the 0 index in the tube list.

I suggest that this is used for unsupported tubes (meaning their conversion factor is not supported in the software yet) and that the factor of this index 0 entry is set to 0.0 to avoid misleading conversions to uSv/h (motto: better have an obviously wrong value like 0.000 uSv/h, than a not obviously wrong value when using a different factor).

Does luftdaten.info do anything special with 0?

rexfue, 2019-11-18:
As far as I know, luftdaten does nothing with special values.

BTW, I now think we can remove the "Radiation unknown" completely. If someone compiles it and uses a new tube, he can (must) define the new tube.

ThomasWaldmann, 2019-11-18:
OK, so let's remove it.

ThomasWaldmann, 2019-11-18:
Hmm, arrays always start at index 0, so we can't remove the entry at index 0...
Also, guess we do not want to shift 1,2,3 to 0,1,2.

So just keep it?

rexfue, 2019-11-19:
What's wrong with changing it? It's only the #define, i guess.
But we can keep it as is, also OK.

ThomasWaldmann, 2019-11-19:
OK, let's keep it as is. It is even somehow useful for playing with unsupported tubes.

And I already have changed the conversion factor to 0.0.

closed 2019-11-19 by ThomasWaldmann