ecocurious2/MultiGeiger #65: fix intro section

ThomasWaldmann, 2019-11-16:
the intro section of the README could be improved / rewritten, typos and grammar fixed.

this was already done once, but reverted on nicola's request, so it is her job now to do that.

ThomasWaldmann, 2019-12-09:
@ecocurious (nicola) do you want to fix that before the workshop?

ecocurious, 2019-12-09:
was denn? ich finds nicht. bin

ThomasWaldmann, 2019-12-09:
einfach das github ticket lesen (siehe top post dort/hier und der link darin).

ecocurious, 2019-12-09:
ja aber auf was konkret der sich bezieht ist dann nicht klar. 

bin krank und werde die nächsten tage nix machen

ThomasWaldmann, 2019-12-09:
wenn du auf den link draufklickst, wird es sehr konkret.

ecocurious, 2019-12-09:
eben nicht, ich soll was fixen aber was genau, ist nicht klar. 
worum gehts genau, kannst du mir das kurz nennen? 

sorry ich bin krank, wenn es nicht wichtig ist bleibts halt so.

ThomasWaldmann, 2019-12-09:
bitte fixe es, wenn du wieder gesund bist. es steht alles da oben.

ecocurious, 2019-12-09:
der screenshot sagt, alles ok
und du weigerst dich zu sagen worum es geht

gute nacht

ThomasWaldmann, 2019-12-17:
OK, typos fixed.

closed 2019-12-17 by ThomasWaldmann