ecocurious2/MultiGeiger #68: counting in 8bits?

ThomasWaldmann, 2019-11-17:
Why does the code count in 8bit wide variables? See `*GMC_count*` variables.

We have a 32bit cpu, the "natural" data type is a 32bit int.

ThomasWaldmann, 2019-11-17:
This seems to work for the low (background) radiation pulse rates, but I guess for higher radiation or more sensitive tubes it could cause malfunction.

rexfue, 2019-11-18:
GMC_counter will never go over 100.

ThomasWaldmann, 2019-11-18:
I guess you assume this because MAXCOUNT is currently 100.

But if pulses are coming in quickly, the ISR will just increment that counter quickly, so even if you have MAXCOUNT 100, it could overflow if the main loop is too slow with catching up.

Also, if one would want to increase MAXCOUNT / the display update interval (see other tickets about this), it is bad to be limited to 255.

Also, is there a special reason? If it was a 8bit uC, I'ld see at least some reason to use an 8bit variable, but on a 32bit uC?

rexfue, 2019-11-18:
Nostalgie - save RAM as effectively as possible :-)

OK, I change it.

ThomasWaldmann, 2019-12-05:
Some issues were fixed, but one was forgotten:
```
         unsigned int  GMC_counts             = 0;
         unsigned char last_GMC_counts        = 0;
```
The latter also needs to be `unsigned int`.

ThomasWaldmann, 2019-12-08:
Fixing this...

closed 2019-12-09 by ThomasWaldmann