ecocurious2/MultiGeiger #70: isr_GMC_counts lost

ThomasWaldmann, 2019-11-17:
In the main loop:
- it first copies the value of `isr_GMC_counts`
- then does **misc. stuff** that takes considerable time and
- then it assigns `isr_GMC_counts = 0`.

`isr_GMC_counts` is incremented by interrupts from the GM tube and all increments that happened while the main loop did **misc. stuff** are lost.

rexfue, 2019-11-18:
You are right, that could lead to a missing pulse. Bad programming, should be changed.
But how fast must the pulse rate be to miss a pulse? In practice, this will not happen.

ThomasWaldmann, 2019-11-18:
If the pulse rate is low (small tube and/or low radiation), it will likely miss only few pulses.

The **misc stuff** takes at least 4ms (just for speaker/led tick, not sure how much the web and other stuff adds to this), it will just miss all pulses that happen in that time.

ThomasWaldmann, 2019-12-05:
@rexfue close?

btw, you can close tickets automatically by appending ", fixes #XXX" to the commit comment of the fix. github then will close the ticket as soon as the changeset is merged into master.

ThomasWaldmann, 2019-12-09:
guess this is fixed by a recent PR.

closed 2019-12-09 by ThomasWaldmann