ecocurious2/MultiGeiger #74: fix ISRs

ThomasWaldmann, 2019-11-18:
Read a comment that arduino's interrupts/nointerrupts()` is not implemented for esp32.

Assuming this is still the case, we need to do it like there (also note the IRAM_ATTR):

https://github.com/espressif/arduino-esp32/blob/master/libraries/ESP32/examples/Timer/RepeatTimer/RepeatTimer.ino

rexfue, 2019-11-18:
Thank you for commenting this. This seems to be really a problem an I will fix it in the next PR. I guess , it should be considered a bug.

ThomasWaldmann, 2019-11-24:
https://github.com/espressif/esp-idf/blob/master/components/freertos/include/freertos/portmacro.h#L210

So it looks like from inside the ISR function, we need to call `portENTER_CRITICAL_ISR`.

From inside main code, we need to call `portENTER_CRITICAL`.

This is required for portability and only on ESP32 both do the same, but this is not required to be the case elsewhere.

ThomasWaldmann, 2019-11-24:
Also, `mux` is an abbreviation for `mutually exclusive` (or `mutual exclusion`), so we need these calls always when accessing a resource shared by ISR and main code - on both "sides".

Only doing it on one side would mean it can not work, because there is no mutual exclusion happening.

ThomasWaldmann, 2019-11-24:
https://techtutorialsx.com/2017/09/30/esp32-arduino-external-interrupts/ they also use it like I mentioned in previous comment.

rexfue, 2019-11-24:
Only for compatibility reasons I agree. And I will put them in. I hope, they are not too expensive.

But i don’t really think it is necessary in our case. If in mainloop interrupt is disabled by portEnter, no interrupt can access our data. If there is an interrupt at the same time, it is pending until mainloop executes portEXIT and then it will be executed. And the other cpu (the second core) never touches our data! So no task switching  corrupts our data.

PR #77 (2019-11-26): Esp32 isr compat
changed  isr routines to be esp32 compliant
deleted all unnecessary 'no interrupt()/interrupt()'
fixed a few global variables
moved attach_interrupt to end of setup()

fixes #74.
closed 2019-11-26 by rexfue