ecocurious2/MultiGeiger #78: mobile logging gps / gmc via bluetooth to android

ThomasWaldmann, 2019-11-23:
@rexfue said bluetooth is relatively easy on esp32 to transfer to an android java app.

good:
- multigeiger in luggage, mobile in your hands
- use storage on mobile
- use gps on mobile
- use internet on mobile

ThomasWaldmann, 2020-02-11:
this (first post) is:
- cheaper than adding a GPS part to the multigeiger
- more comfortable then dealing with microSD cards (that could be used at the esp32 for logging)

t-pi, 2020-03-10:
First tests with BLE showed a partition problem of the ESP32: The size was way below the 4MB flash, yet an error was thrown:
```
Error: The program size (1798523 bytes) is greater than maximum allowed (1310720 bytes)
RAM:   [==        ]  20.6% (used 67372 bytes from 327680 bytes)
*** [checkprogsize] Explicit exit, status 1
Flash: [==========]  137.2% (used 1798523 bytes from 1310720 bytes)
esptool.py v2.6
```
[This link](https://github.com/platformio/platform-espressif32/issues/93) refers to the standard arduino partition tables which need to be modified. The code now compiles with 
```
board_build.partitions = no_ota.csv
```
- however, we want OTA. 

Do you guys know how to adapt the [partition tables](http://docs.platformio.org/en/latest/platforms/espressif32.html#partition-tables) for our OTA to work and still have ~1.7-1.8 MB for our binary?

ThomasWaldmann, 2020-03-10:
The arduino ide offers a partitioning scheme "minimal spiffs" that offers that.

t-pi, 2020-03-11:
Looks good: 
```
Flash: [========  ]  84.5% (used 1661219 bytes from 1966080 bytes)
```
(I also changed to the espressif BLE libs...)
Thanks!

rexfue, 2020-03-12:
Our board (Heltec Wifi Kit 32) has 8MB, so plenty of memory. But both platformio and arduino ide say, this board only has 4MB.
I tried to change the board.txt file, but with no success.
If anybody knows, how to tell platformio or Arduino ide, that we have 8MB , please tell us.

t-pi, 2020-03-12:
Hmm, did you try to edit this file?
...\.platformio\platforms\espressif32\boards\heltec_wifi_kit_32.json

rexfue, 2020-03-12:
Thank you - that worked. I always tried the different 'board.txt' files :-(

These are the changes to do in heltec_wifi_kit_32.json:
--- a/heltec_wifi_kit_32.json_org
+++ b/heltec_wifi_kit_32.json
@@ -9,6 +9,7 @@
     "f_flash": "40000000L",
     "flash_mode": "dio",
     "mcu": "esp32",
+    "partitions": "default_8MB.csv",   //<<<<<<<<<<<<<<<<<<<<<<
     "variant": "heltec_wifi_kit_32"
   },
   "connectivity": [
@@ -23,9 +24,9 @@
   ],
   "name": "Heltec WiFi Kit 32",
   "upload": {
-    "flash_size": "4MB",       // <<<<<<<<<<<<<<<<<
+    "flash_size": "8MB",      // <<<<<<<<<<<<<<<<<
     "maximum_ram_size": 327680,
-    "maximum_size": 4194304,   // <<<<<<<<<<<<<<<
+    "maximum_size": 8388608, // <<<<<<<<<<<<<<<<<
     "require_upload_port": true,
     "speed": 460800
   },

Caution: It forgets the Wifi credentials, because of the new partition scheme. But that happens only the first time.

t-pi, 2020-05-05:
Question: 
Current flash size comparison of latest dev-build vs. BLE shows strong increase to >50%.
Will this be a problem for OTA-Updates?

```
v1.13.1"a"-dev from 06.05.2020
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=         ]  14.3% (used 46728 bytes from 327680 bytes)
Flash: [===       ]  31.7% (used 1059767 bytes from 3342336 bytes)
v1.13.1-dev-ble based on v.1.13.1-dev of ~28.04.2020
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [==        ]  20.4% (used 66740 bytes from 327680 bytes)
Flash: [=====     ]  50.8% (used 1698231 bytes from 3342336 bytes)
```

t-pi, 2020-05-05:
Hmm, and there doesn't seem to be an easy way to strip down BLE:
https://github.com/nkolban/esp32-snippets/issues/336#issuecomment-354659830
(size of basic app with BLE for ESP32 lib ancestor in 2018 at ~900 kB)

ThomasWaldmann, 2020-05-06:
If you set up your arduino IDE like described in current multigeiger docs, you'll have a layout for the smallest flash size we support (4MB) and OTA partition layout. Then you can directly see whether you get close to 100%.

ThomasWaldmann, 2020-05-16:
@t-pi just released V1.14.0! So feel free to PR BLE stuff now. :-)

t-pi, 2020-05-16:
Great, thanks! Will adapt to current source and PR, hopefully this WE!

commit f7363918 (2020-05-22): BLE connectivity (#311)

add bluetooth (BLE) support, fixes #78

- implement BLE heart rate service with notifications about current CPM rate
- use "unused" switch for "ble_on"
- allow configuration via webconf
closed 2020-05-22 by ThomasWaldmann