ecocurious2/MultiGeiger #93: Docs Case / Deployment

ThomasWaldmann, 2019-12-06:
Is there any docs already about assembling into the case / outside deployment?

Rest in german because it refers to german docs:

Das sind die relevanten Teile von der Liste im Repo:
- 1 Installationsrohr starr M40 grau 3m EAN 8027788360043 www.hornbach.de
- 0.25 Tehalit Kabelkanal 20x35 mm reinweiß 2m EAN 4012740327541 www.hornbach.de
- 0.1 Silikon-Kleber   Fix All Crystall 80 Gramm EAN 5411183085084 www.hornbach.de
- 2 aponorm® Schraubdeckel-Dose 30 g / 35 ml  ??? Apotheke / Pharmacy

Fragen:
- Installationsrohr - braucht man die ganzen 3m oder auf ne bestimmte Länge zuschneiden (auch für Transport relevant)?
- Kabelkanal - was macht man damit genau?
- 2 Dosen - die sind wohl zum Abschliessen des Rohrs + Kleber.

ThomasWaldmann, 2019-12-06:
Auch bei Hornbach zu bekommen, vielleicht auch interessant?:
- Klemmschelle M40 für Installationsrohr grau 5 Stück, EAN 9002579842907
- Kaiser Dichtstopfen M40 104040, EAN 4013456536852

ThomasWaldmann, 2019-12-07:
@WwwBoehriDe kannst mir mal bitte die (gekürzten) Längen sagen für...?
- Inst.-Rohr M40
- Kabelkanal

Frage fuer einen Velomobilfahrer, der das transportieren muss. :-)

WwwBoehriDe, 2019-12-07:
Hallo Thomas,
Oberes  Rohr:        850 mm
Unteres Rohr:        850 mm, schräg abgesägt (ca. 45 Grad) (Längster Teil ist 850 mm lang)
Installationsplatte (Teil vom Kabelkanal): 720 mm

ThomasWaldmann, 2019-12-07:
Danke, hat reingepasst! :)

ThomasWaldmann, 2019-12-07:
Gibt's zum Aufbau irgendwo ne Zeichnung / Beschreibung?

WwwBoehriDe, 2019-12-07:
Hi, leider noch nicht - bin gerade dabei eine Anleitung zu machen. Nächstes Wochenende brauchen wir die eh.

ThomasWaldmann, 2019-12-07:
OK. Ich bastel morgen evtl. im shack wieder etwas weiter.

Wenn Du Deine Anleitung vorher veroeffentlichst, kann ich den Beta-Test davon machen. ;-)

So grob ist mir das schon klar, es geht primaer um die Abmessungen, Höhe Zählrohr über Boden und sonstige Feinheiten.

Wenn in die Doeschen Trockenmittel SilicaGel rein soll: ich glaub das fehlt auf der Stueckliste.

WwwBoehriDe, 2019-12-07:
Alles klar. Ich schaue dass ich es Dir heute Abend (bzw. morgen früh) per Mail schicke. 

Die Idee ist es einfach offen zu betreiben - ohne Trockenmittel. Aber Du wirst sehen!

ThomasWaldmann, 2019-12-21:
Case / Deployment docs are there:

https://github.com/ecocurious/MultiGeiger/tree/master/docs/Bauanleitung

So this is fixed.

closed 2019-12-21 by ThomasWaldmann