ecocurious2/MultiGeiger #97: speaker / led tick not working any more

ThomasWaldmann, 2019-12-08:
with current master branch: no ticks, no flashing.

cpm display seems to work though.

ThomasWaldmann, 2019-12-08:
50839d008db253d3b1d55e3c9f7b2a1e27f2035c works
377db2a59d9f45ab155a64c2d7cc7b132e597d22 does not work

bisecting this now ...

ThomasWaldmann, 2019-12-08:
git bisect tells this:
```
b9bb9bc5423bb28500f56e9dd736c133e044411c is the first bad commit
commit b9bb9bc5423bb28500f56e9dd736c133e044411c
Author: Reinhard X. Fürst <rexfue@gmail.com>
Date:   Tue Nov 19 16:16:03 2019 +0100

    isr routines now esp32 conform
    removed all "nointerrupt()/interrupt()"
    fixed a few global variables
    changed char variables to int

:040000 040000 128d05f015e1df93ae5ece922024ed7683d5a3ee fbe842465ff0a6e60b28b193b6f42e1368ea5fa7 M	multigeiger
```

rexfue, 2019-12-08:
Was bedeutet das? Was ist bisect?

ThomasWaldmann, 2019-12-08:
automatisiertes finden des commits, ab dem es nicht mehr tut. macht intern ne binaere suche, d.h. selbst bei 1000 commits hat man den "schlechten" nach 10 versuchen.

rexfue, 2019-12-08:
Ah, OK, danke. Kann Du da weiter suchen? Ich komm da grad nicht dazu.

rexfue, 2019-12-08:
What about the switch? SW1 = OFF(1) disables the speaker tick.

ThomasWaldmann, 2019-12-08:
I don't have switches on this geiger. Also userdefines.h is correct and it does tick with an old sw version, just not with latest one.

Yeah, I have some time, will work on this.

rexfue, 2019-12-08:
OK thanks.

commit 449b6099 (2019-12-08): Merge pull request #99 from ThomasWaldmann/fix-ticking

fix not working speaker and led tick, fixes #97
commit 840500bf (2019-12-08): fix not working speaker and led tick, fixes #97
closed 2019-12-08 by ThomasWaldmann