ecocurious2/MultiGeiger #105: tag vs release discrepancy?

ThomasWaldmann, 2019-12-17:
https://github.com/ecocurious/MultiGeiger/releases

there is a 1.11 vs. a 1.11.01.

Seems one has a release made, the other one has no github release (only shows up because there is a tag).

rexfue, 2019-12-17:
Maybe I did it the wrong way. Release 1.11 sits now on tag 1.11.01.
After making the release, i had to change a little thing (name of luftdaten server), so that Jürgen can programme the right thing.

ThomasWaldmann, 2019-12-17:
Well, always keep release tags and displayed version in sync.

That would mean if you only want to use x.y and not x.y.z, you need to do use that everywhere.

Version numbers are for free. :)

rexfue, 2019-12-17:
OK, thanks.

ThomasWaldmann, 2020-01-09:
@rexfue so, do you intend to fix that or will it be just superseded by some future release?

rexfue, 2020-01-09:
I prefere to supersed it in next release.

ThomasWaldmann, 2020-01-13:
1.11 commit: 5304e3114d82abbf981ae63e821de4f483991c48

1.11.(0)1 commit: a941ac8c990292f77a4a0b511640b1bca59eb2dc

release page: strange that github displays 1.11 above 1.11.(0)1 although it is older

ThomasWaldmann, 2020-01-13:
I'll fix it...

ThomasWaldmann, 2020-01-13:
Fixing it involved simplifying the version numbers: V1.11.0 and V1.11.1 - that sorts correctly.

The previous style did not sort correctly (so just do not put dates in there, they are already in the changelog, in the geiger source, in git, so we do not also need them in the tag).

closed 2020-01-13 by ThomasWaldmann