ecocurious2/MultiGeiger #106: RFC: idea: connect a WS2812 LED

ThomasWaldmann, 2019-12-17:
If the geiger is deployed outside in its grey plastik casing, there is little use for:

- the small onboard LED (pin 25 on all controllers)
- onboard OLED display
- maybe also the onboard piezo (except if one is very close with ones ears)

So, what remains is whatever it publishes on the internet.

For the geiger operator (and everybody else close-by) it would be nice to have a more direct display / feedback mechanism - this is how the WS2812 idea came up.

These LEDs are RGB programmable and exist as single LEDs (I got some 8mm ones) and also as LED strips (all with only 1 data input). Links (only to illustrate):

- https://www.sparkfun.com/products/12877
- https://www.sparkfun.com/products/12661

They need: Vcc (5V), GND, DATA_IN (usually works with high level from 3.3V output)
They provide: DATA_OUT for next LED, if any

So, for outside geigers, we could connect Pin25 to one WS2812B LED and let it shine in whatever colour or pulse pattern we like. So the geiger would have a nice shiny upper end. :-)

Mounting options:

- the (single) LED could stick through a whole in the upper end cap and get sealed by the glue that gets put on there anyway. The end cap would stay inside the grey tube and only the LED would stick out a little at the top.

- maybe even more pretty / noticable: let the white end cap stick out from the grey tube a bit and keep the single LED completely inside, so it will illuminate the end cap as a whole from the inside. the glue sealing would only happen on the side of the end cap in that case (not on the top). no drilling required.

The required software should be readily available via a library, these LEDs are quite popular - so our part is quite easy.

What do you think about it?

ThomasWaldmann, 2019-12-17:
(I already used such strips from micropython, was quite easy.)

ThomasWaldmann, 2019-12-17:
https://github.com/adafruit/Adafruit_NeoPixel

Not sure if it helps much for 1 LED, but for a whole strip, this might be less expensive on the CPU as it uses the RMT hardware:

https://github.com/bertmelis/esp32WS2811

rexfue, 2019-12-18:
Really funny idea. I know these RGB devices and use them at the moment in my 'word clock'.
But I don't think that people want to have a flashing or colourful lighting stick in their garden.
So you are welcome to build and implement this, but we won't do it in our project - IMHO.
You can switch off LED, speaker and display by software or by the DIP switch if you populate the board with it.

ThomasWaldmann, 2020-01-19:
Pinout:
- Din
- +5V
- GND (long pin)
- Dout

(need to verify this myself first)

ThomasWaldmann, 2020-02-02:
BTW, I did some first experiments:

Seems like some simple status showing code + the `NeoPixelBus` library would add only about 5kB to our compiled code size and it can be switched on/off via some `#define STATUS_LED 0/1` in `userdefines.h`.

Also, the end caps we use are semi transparent, so they have a nice glow when they have such a 8mm LED inside.

rexfue, 2020-02-02:
I measured the level at pin 25: It's  3.3V, so that shouldn't be a problem.

ThomasWaldmann, 2020-02-02:
OK, now locating the pins to connect the WS2812 LED to:

```
o o o o o o o o o o ...
Prg
USB                             OLED
Rst
o o o o o o o o o o ...
1 2 3 4 5 6 7 8 9 10 ...

1 == GND
2 == +5V
10 == LED
```

ThomasWaldmann, 2020-02-02:
OK, it somehow works, mostly.

Guess I need some serial damping resistor (to avoid reflections) or a capacitor (to stabilize voltage) at the LED or a thicker 5V wire, sometimes it shows wrong colors.

ThomasWaldmann, 2020-03-05:
@rexfue said 13 and 17 are free on current hardware.

jensliebau, 2020-04-07:
> Really funny idea. I know these RGB devices and use them at the moment in my 'word clock'.
But I don't think that people want to have a flashing or colourful lighting stick in their garden.
So you are welcome to build and implement this, but we won't do it in our project - IMHO.

If you place a multigeiger in your garden close to a window you can hear "speakerTick". On the pro side it easy to detect power failure, but depending on how close it is placed to your window it's a little annoying, too. Now than we can disable "speckerTick" in the upcoming release, it would be nice to have some LED for monitoring for state and power failures.

-> So I would vote for adding LED support in an future release.