ecocurious2/MultiGeiger #109: CR/LF on serial output

rexfue, 2019-12-21:
If we send  only \n in the log outputs (for example: char  *Serial_Logging_Body   = "%10d %15d %10f %9f %9d %8d %9d %9f %9f\n";), Windows terminal programs have problems: they expect \r\n by default.
So I think, we should change every \n into \r\n. The Arduino "Serial.println" also sends both.

ThomasWaldmann, 2019-12-21:
Well, at least this needs testing on Linux first.

rexfue, 2019-12-21:
tested on Linux with minicom - no problem.

Yes, the sprintfs make the problem. You and me didn't see the problem because you use linux and I use MAC and both are happy with only LF. And no one else looked at the serial output, I guess.
I will change that and build a PR the next days (after Christmas). So you can check and test.

ThomasWaldmann, 2020-01-01:
@rexfue guess this was fixed by PR #116 ?

rexfue, 2020-01-01:
Yes. I close it

closed 2020-01-01 by rexfue