ecocurious2/MultiGeiger #110: docs questions

ThomasWaldmann, 2019-12-22:
Da es die deutschen Docs sind, Fragen auch auf deutsch.

Ich aktualisiere / fixe grad die Doku (also bitte jetzt keine Aenderungen im Dokument, sonst kriegen wir das nicht gemerged).

## Was hat der Filename `Aufbauanleitung_2019_13_09.odt` zu bedeuten?

Ist das ein Datum? Wozu?

Wir haben git fuer die Versionskontrolle, man muss da kein Datum und auch keine Versionsnummer reinschreiben.

## Im Dokument, Fusszeile

```
Multigeiger-Version  V1.4 jb/rxf 2019-09       Handout-Version_1/19        Lizenz:  CC BY-NC-SA 3.0 DE
```

Fragen:
- `Multigeiger-Version  V1.4 jb/rxf 2019-09` ist damit die Hardware-Version gemeint? Können wir es dann Hardware-Version nennen. Mutigeiger ist ja alles, Hardware+Software+Docs.
- `Handout-Version_1/19` - was bedeutet die 1, was bedeutet die 19?
- muss die Lizenz in jeder Fusszeile stehen oder reicht das auch einmal oben im Dokument?

rexfue, 2019-12-22:
Anmerkung zum Datum in diesem Dokument: Wenn sich das jemand runterlädt und ausdruckt oder er es ausgedruckt bekommt, macht es m.E. schon Sinn, dass da ein Datum (wie und wo auch immer) drin steht.

ThomasWaldmann, 2019-12-22:
Bin ich voll Deiner Meinung. Das meinte ich aber nicht, s.o..

ecocurious, 2020-01-02:
hi thomas, 
frage 1: kannste ändern in hardware, bitte dann auch pdf neu generieren und hochladen 
frage 2: bedeutet version 1/2019 
frage 3: macht sinn und stört mich nicht

ThomasWaldmann, 2020-01-02:
1: ok, aendere ich in "Hardware-Version".

2: 1 == Januar oder 1 == erste Version?

3: ok, mir egal, ich fand's eben etwas "overkill" und die fusszeile ist eh recht voll.

ThomasWaldmann, 2020-01-16:
1 == erste Version.

ThomasWaldmann, 2020-05-22:
As these docs were now committed in time order, there's nothing to do left except to remember:
- do not put dates / version numbers into filenames
- use unambiguous versions / dates (if you must) inside the document

I changed the doc footer in Aufbauanleitung to be clear that it is the HW Version and that the docs version is a ISO-like date.

commit 510e9e30 (2020-05-23): Merge pull request #339 from ThomasWaldmann/fix-docs-unclear-version-date

fix unclear version / date in Aufbauanleitung, fixes #110
commit 8256b40f (2020-05-23): fix unclear version / date in Aufbauanleitung, fixes #110

- "HW-Version" is more clear than "Multigeiger-Version"
- Changed the timestamp from 2020-01 to 2020-MM-DD, so it
  can be used in a more clear way next time the doc is updated.
closed 2020-05-23 by ThomasWaldmann