ecocurious2/MultiGeiger #112: PCB too wide for tube case

ThomasWaldmann, 2019-12-22:
I noticed that when trying to put the 1.4 PCB into the plastik tube, it almost fits, but not quite.

So guess the PCB should be cut some mm less "high" than it is now to fit into the inner tube width.

Also, some components hit the tube casing from the inside, e.g. the inductivity and the microcontroller if the PCB is forced into the tube. This could be resolved by having the PCB not sit in the middle of the tube, but slightly in the less wider part, so there is more space above the PCB.

Nordrunner01, 2020-01-02:
Moin

Bez. auf: https://github.com/ecocurious/MultiGeiger/blob/master/docs/Bauanleitung/Aufbauanleitung_2019_13_09.pdf Teil3, Seite 8

Was für ein Rohr ist das denn? HT Abflußrohr? 
Maßangaben in welcher Größe? (Millimeter)

rexfue, 2020-01-02:
Masse in mm. Das ist ein Elektriker-Installation-Rohr mit 40mm Durchmesser. Abflussrohr ist etwas dicker -> etwas schlechter.

rexfue, 2020-01-02:
Stimmt net, sind cm!

Nordrunner01, 2020-01-02:
Hmmm....
Das wird nicht lange halten. Nicht UV-Stabil.

Oder gibt es gegensätzliche Erfahrungen?

rexfue, 2020-01-02:
m.E. noch nicht. Aber Du hast sicher recht, das ist nicht UV-stabil. Jürgen ist schon auf der Suche nach einer Alternative.

ThomasWaldmann, 2020-01-05:
Bitte die UV und Rohrmaterial-Diskussion in ein separates Ticket verschieben.

Hier geht es darum, dass die Platine schlecht in das Rohr passt.

DeeKey, 2020-01-05:
Where is PCB itself? Is it opensource?

ThomasWaldmann, 2020-01-05:
@DeeKey see `docs\hardware\`.

rexfue, 2020-01-09:
I cannot follow your complaining about pcb not fixing into tube. My pcb and all the pcbs in the workshops fit fine into the tube. Maybe your sockets are the wrong ones and too high? 
So for me there's no reason to redesign.

ThomasWaldmann, 2020-01-09:
@rexfue Can you precisely measure the inner diameter of your plastik tube?

ThomasWaldmann, 2020-01-16:
I measured the tube I bought: inner diameter 37.4mm, outer diameter 40.4mm.

ThomasWaldmann, 2020-01-16:
Problem solved:

The old PCB version I have expected that one needs to break off another ca. 3-4mm stripe (besides breaking the 3 PCBs to separate them from each other).

After breaking that stripe off, it'll fit easily.

The newer PCB production does not need the stripe to be broken off, the stripe is not present.

closed 2020-01-16 by ThomasWaldmann