ecocurious2/MultiGeiger #114: Bezugsquelle für den Bausatz?

Nordrunner01, 2019-12-26:
Moin.

Ich täte mich für für die Bausätze interessieren um eventuell hier oben bei uns ( https://nucleon-ev.eu/ ) ähnliche Informationsveranstaltungen durchführen zu können und, ähnlich wie beim luftdaten-Projekt Messstellen zu generieren.

Ein SBM-20 Zählrohr hätte ich noch liegen. Eine ODL-Messstelle ist auch nicht weit weg.

Was haltet ihr von der Möglichkeit das Interessenten die Platinen selbst beschaffen zu können? Ähnlich wie beim PiGi: https://pcbs.io/share/r1Dq6
Oder habe ich diese Möglichkeit noch nicht gefunden/überlesen?

Gruß

rexfue, 2019-12-26:
Hi, Bauteileliste und Eagle-Files der Platine sind hier im GitHub. Bausätze verschickt auf Anfrage auch Kollege Jürgen (juergen@boehri.de). Wäre schon, wenn es da 'oben' bei Euch auch bald ein paar Zähler gäbe.

Nordrunner01, 2019-12-26:
Moin @rexfue 

Es ist nicht meine Art, mich mit fremden Federn zu schmücken. PCBs.io gewährt einen gewissen Rabat, wenn Platinen bestellt werden. Diese kommen dem Einsteller zu gute und werden bei eigenen Bestellungen verrechnet.
Von daher würde ich die Variante bevorzugen, wenn Jürgen diese dort selbst einstellt und auch den Nutzen davon hat.

Und ja, es wird Zähler geben. Allerdings mit Erfassungsplatinen vom Ali bisher. Datenübertragung erfolgt mit LoRa an unsere LoRaWANGateWays hier oben. Ich glaube das Projekt heist Opengeiger-TTN. Eigentlich ist es, glaube ich, das selbe wie hier.

Gruß

rexfue, 2019-12-26:
Ja, das ist natürlich eine Preisfrage. Eine kurze Anfrage hat ergeben, dass die ca. 4x teurer sind als JLCPCB, wo ich die Platinen bestelle.
TTN habe wir auch auf dem Schirm bzw. das geht auch schon - mit der entsprechenden Hardware. Wir bauen gerade ein Gateway von TTN nach luftdaten.info (bzw. neu: sensor.community), damit alle unsere Zähler (egal ob LoRa oder WLAN) auf unserer Karte angezeigt werden können.

Nordrunner01, 2019-12-26:
Ok.
JLCPCB haben wir auch auf dem Schirm, aber noch nichts gemacht. Ich fand das Konzept mit der Rückvergütung eben nicht schlecht.

Das Thema Karte, wäre aus meiner Sicht einen eigenen Issue wert. :-)

Gruß

PS: Mail an Jürgen ist raus.

ecocurious, 2020-01-02:
Hallo Nordrunner, 
Ziel ist es, das unsere Messtationen alle die gleiche Hardware haben, damit die Daten vergleichbar sind. Darum wäre es gut, wenn du unsere Bauteile komplett verwendest. Und auch unser Gehäuse, denn die Wandstärke des Gehäuses spielt eine Rolle (zu dick= mehr Abschirmung). Wir machen gerade eine Liste, weil es sehr viel nachfrage gab, nachdem wir auf dem 36c3 waren. Wenn du Interesse hast, einen Bausatz zu bestellen, dann schreib mir an we@ecocurious.de ich setze dich mit auf die Liste. wir machen eine große Sammelbestellung und schicken dann alles an Euch raus. Liebe Grüße von Nicola, Projektleiterin

Nordrunner01, 2020-01-02:
Moin Moin Nicola und ein gesundes, neues Jahr.

Ich hatte mich wegen eines Bausatzes, oder dessen Buzugsmöglichkeit schon in Verbindung gesetzt.

Und ja, ich möchte auf diese Liste!

Bei uns hier oben im Norden gibt es durchaus Loide, die sich für das Thema open-Geiger und einem einheitlichem  Messstellen Netzwerk interessieren.
Ich teile die Ansicht von gleichartigem Aufbau der Messstellen/Sensorik.

Wir (Nucleon e.V. und IoTUsergroup.de) sind in vielen Bereichen interessiert. Der Nucleon e.v. unterstützt den https://codefor.de/schleswig_flensburg/ .

Ich frage mich, ob man die OK-Labs, mit ihren besonderen Zielsetzungen für ein großes Gesamtprojekt unter einen Hut bringen könnte. Die Überschrift könnte Umweltdatenerfassung lauten, die dann in eine große Datenbank fließen und welche dann für Analyse angezapft wird.

Interessant ist ja auch das Projekt luftdaten.info, die gerade an einer Erweiterung bauen, welche die Erfassung von Lärm betrifft.

Ich meine, das ecoqurious doch aus der selben Ecke kommt. Ist da eine Zusammenarbeit geplant?

Ich horche mich mal um, wer noch interesse hat an eurem Projekt teilzunehmen.

Gruß

ecocurious, 2020-01-02:
ich schreib dir gerade ne Mail :-)

ThomasWaldmann, 2020-05-21:
Can we put some official answer to that question into the docs?

ThomasWaldmann, 2020-05-22:
We can either close this ticket (meaning we do not publish how to get kits) or add it to documentation.

@WwwBoehriDe any idea what should go into docs?

WwwBoehriDe, 2020-05-22:
Hi,
ich denke das sollten wir in der nächsten TelCo besprechen. Ich denke schon dass es gut wäre wenn die Leute uns kontaktieren könnten. Von dem her wäre ein Hinweis dass es Bausätze gibt gut.

ecocurious, 2020-05-23:
Bestellung von Bauteilen: Liebe Leute, sobald Corona es zuläßt, werden wir wieder Workshops anbieten, wo du gemeinsam mit uns den stationären Multigeiger und das Outdoor-Gehäuse zusammenbauen kannst. In der Zwischenzeit können wir dir aber auch gerne einen Bausatz schicken. Schreib eine Mail an we@ecocurious.de UND juergen@boehri.de, wir schicken Dir dann alle weiteren Infos per Mail.

WwwBoehriDe, 2020-05-23:
Hallo,
ich habe extrem schlechte Erfahrungen damit gemacht wenn eine Mailadresse als Klartext im Netz steht. Daher wäre es mir recht wenn man, zumindest bei meiner Adresse, das so ändern könnte:

... UND Jürgen Böhringer (Mailadresse: siehe www.boehri.de, bei Impressum), wir schicken ...

... vielleicht kann man bei Nicola ja das so ähnlich machen. Das Problem ist dass ansonsten Bots die Mailadresse finden und zuspammen.
    Viele Gruesse

ThomasWaldmann, 2020-05-24:
Da ihr spezielle Anforderungen habt, schreibt bitte den Text selbst fertig, so dass ich ihn nur noch per Copy und Paste übernehmen muss.

Die Doku ist auf Englisch und Deutsch (Uebersetzung kann ich machen).

Aber wenn ihr z.B. nur in bestimmte Länder liefern wollt, den Text bitte dementsprechend selbst machen.

Sagt einfach Bescheid, wenn ihr fertig seid. @WwwBoehriDe @ecocurious

ThomasWaldmann, 2020-06-06:
ping.

ThomasWaldmann, 2020-06-13:
Aus ner Mail von Nicola:

Wir schreiben es so: 

Bestellung des Multigeiger-Bauteile-KIts: 

Liebe Leute,

sobald Corona es zuläßt, werden wir wieder Workshops anbieten, wo Du gemeinsam mit uns den stationären Multigeiger und das Outdoor-Gehäuse zusammenbauen kannst.

In der Zwischenzeit können wir dir aber auch gerne einen Bausatz schicken. Schreib eine Mail an Nicola (Mailadresse: siehe ecocurious.de, bei Kontakt) UND Jürgen ((Mailadresse: siehe www.boehri.de, bei Impressum), wir schicken Dir dann alle weiteren Infos per Mail.

ThomasWaldmann, 2020-06-13:
Anm.: Ich wuerde nen FAQ/Doku-Eintrag nicht mit "Liebe Leute," anfangen.

@WwwBoehriDe ok so?

ecocurious, 2020-06-15:
ok wie?

ThomasWaldmann, 2020-06-15:
Mal an alle:

Wenn ihr ne Benachrichtigung von Github kriegt, dann klickt bitte auf den darin enthaltenen Link zu Github und lest Euch es auf der Web-Oberflaeche durch und fragt nicht einfach via E-Mail Reply "wie, wo, was, warum?".

Spart allen viel Zeit, Euch auch.

Danke!

ThomasWaldmann, 2021-01-10:
see also #367.

ThomasWaldmann, 2021-06-18:
shifted to next milestone as i did not see anybody actively working on this.