ecocurious2/MultiGeiger #117: Wäre dieses Teil eine weitere Möglichkeit als Prüfstrahler

Nordrunner01, 2019-12-30:
Moin.

Ich wühle mich gerade durch die Doku von www.opengeiger.de .

In meinem Dunstkreis gibt es einige Loide, die sich auch für das Thema interessieren. Darum haben wir uns als 1., günstige Möglichkeit für dieses Produkt [www.banggood.com Assembled-DIY-Geiger-Counter-Kit-Module-Miller-Tube-GM-Tube-Nuclear-Radiation-Detector](https://www.banggood.com/Assembled-DIY-Geiger-Counter-Kit-Module-Miller-Tube-GM-Tube-Nuclear-Radiation-Detector-p-1136883.html?rmmds=buy&cur_warehouse=CN) entschieden und bestellt.

Dabei wird dann, weiter unten vom Angebot ein sog. "Ion Chamber Metal Geiger Counter Check / Test Source Fire Alarm Security System Source Smoke Tester Sensor" angeboten.
Laut Beschreibung:
This sensor contains tiny amount of Americium 241 isotope.

Americium emits alpha particles and weak gamma radiation, alpha particles completely blocked by the casing of the sensor, gamma radiation however freely passes through the casing but due to very low energy around 60 keV quickly absorbed by air (maximum run through air 15cm).

It is a great source for checking gamma only sensitive Geiger counters, if you have such type of counter it will be impossible to test it with other sources which emit beta particles only.

It could be used to test gamma spectrometers as well.

Add a nice "hot" piece to your collection which emits alpha and gamma radiation.

SBM-20 tube based Geiger counter indicates 140 Counts per minute from this source with gamma filter applied (approximately 80 uR/h or 0.8 uSv/h accordingly).

Specifications:

Ionization source characteristics and radiation safety performance:

The ionization chamber fitted with a high performance and low activity of Am-241 ionization

Ionization source activity: 0.5uCi ( 18KBq ) 0.8uCi±10% ( 30KBq±10% )

Ionization source a spectroscopy: peak: 4.5MeV ± 10%, FW HM<0.7MeV

Safety grading standards ( GB4075, IS02919 ) C64444

Ionization chamber radiation dose rate at 25cm: 0.03mGy/ (1mGy value lower than population dose standard)

Also für einen Preis von unter 3 €ronen, wäre es doch ein günstiger Prüfstrahler. Oder liege ich da falsch.

Die nächste Frage, wäre, kann man es überhaupt in die BRD einführen? :-)

Gruß

ThomasWaldmann, 2020-01-01:
https://en.wikipedia.org/wiki/Americium-241
https://de.wikipedia.org/wiki/Americium

Ich bin da kein Experte, aber nen starken Alpha-Strahler (der Gamma nur als kleinen Nebeneffekt generiert) als Testsource fuer ein Gamma-Zaehlrohr zu nehmen kommt mir abwegig vor.

ThomasWaldmann, 2020-01-01:
Div. mehr oder weniger harmlose und einfach beschaffbare Strahlungsquellen:
- Pottasche (oder generell alles, was grossteils aus K besteht und deshalb auch K-40 enthält)
- Glühstrümpfe (enthalten Thorium)
- "Depression Glas" / Glasperlen (mit Uran blaugrün angefärbt)
- Leuchtfarbe (die, die dauernd etwas leuchtet)
- Fiesta-Ware (Keramik aus den 60er Jahren, mit orange-farbiger Uran-haltiger Glasur)
- div. Uran- oder Thorium-haltige Mineralien

Nordrunner01, 2020-01-02:
Ich fand diese Aussage hilfreich:
SBM-20 tube based Geiger counter indicates 140 Counts per minute from this source with gamma filter applied (approximately 80 uR/h or 0.8 uSv/h accordingly).

Und von daher geeignet.

Trotzdem Danke für die Zusammenfassung.

closed 2020-01-02 by ecocurious