ecocurious2/MultiGeiger #120: Firmware Updates

ThomasWaldmann, 2020-01-05:
## Need for a practicable firmware upgrade procedure

### Context
- MultiGeiger likely sits in a sealed tube and the USB port can't be accessed easily without disassembling the device / breaking the seal / resealing it.
- We need to also consider users who are not very technical, so doing manual upgrades (no matter how) might be an issue.
- Guess we would like to keep the devices up-to-date with the latest stable compatible firmware version (for bug fixes and features).

### Upgrade options
Considering this, manual upgrades via USB cable or manual via WiFi using a browser are no good options.

What's left are automated upgrades via WiFi from a internet server.

### Ideas

If the update file is cryptographically signed by a release manager, its integrity and its authenticity can be verified after it is downloaded, before it is flashed (it could be even verified again after flashing [reading it from flash] before switching to it to verify the flash procedure).

That way, the security of the transmission channel or of the server does not matter at all (as long as the signing key[s] are kept secret). We would never flash a tampered firmware.

Also, as long as the device sometimes manages to correctly transfer the file, it would work ok. We would never flash a corrupted firmware (but just retry later).

## Priority

Rather high as it is a pain to upgrade the devices by other means (see "Context").

ThomasWaldmann, 2020-01-05:
## Implementation ideas

### ESP IDF based

Does this mix with Arduino stuff?

https://docs.espressif.com/projects/esp-idf/en/latest/security/secure-boot.html#signed-app-verification-without-hardware-secure-boot 

This pretty much is what we want, I guess.

### Arduino based

WwwBoehriDe, 2020-01-07:
There was the idea to seal the device completly using silica gel to reduce humidity. Currently it looks like if this is not necessary. So the current housing-design is extremly easy to open and not sealed at all. So you have access to the device and its Micro-USB-Connector in just some seconds.

Please check:
https://github.com/ecocurious/MultiGeiger/blob/master/docs/Bauanleitung/Aufbauanleitung_2019_13_09.pdf
Page 8

I think Reinhard has alredy looked into OTA and found out that there are different memory sizes available on the Heltec-PCBs, which is problematic for OTA. But anyway. Probably its worth to check it again.

ThomasWaldmann, 2020-01-08:
Even if you can reach the USB plug, it is still uncomfortable that way (and unlikely that all/most people frequently do that). Of course one does not need to install all the updates, but maintaining a lot of differently outdated or buggy versions likely is also not fun for the developers.

Reinhard told me that the overall code size might be problematic (see also #84 about that) for the 4MB flash size (but not for the 8MB devices) when using dual partitions for OTA.

ThomasWaldmann, 2020-01-09:
Arduino's partition scheme "Minimal SPIFFS" offers 1.9 MB max. "APP" partitions (two of them, for OTA updates) and 190kB SPIFFS.

@rexfue Some questions:

- do we use a significant SPIFFS amount or are 190kB sufficient?
- what is the maximum binary size / app size you've seen yet? I've seen 0.8MB, but I do not have LoRa.

BTW, our own code is no way that large, most of that are libraries and likely that part could be optimized (not sure if that can be done with Arduino framework though).

rexfue, 2020-01-09:
We don't use SPIFFS (at the moment)
Currently program size is a little bit more than 1 MB in both versions, so this partition scheme should work

ThomasWaldmann, 2020-01-09:
I just wondered where it stores the data one enters via the web interface.

rexfue, 2020-01-09:
It will be saved in "NVS" (or "preferences"), starting at 0x9000 and 0x5000 long (this depends on partition scheme). So its enough for our parameters.

ThomasWaldmann, 2020-01-11:
OK, I'll now work on implementing a simple solution based on OTAWebUpdater.

It won't be very secure, but the point is just having anything that does not require plugging into USB.

This won't be the desired state of OTA for MultiGeiger, but a start.

It should be superseded by a server based update mechanism with cryptographically signed firmware.

ThomasWaldmann, 2020-01-12:
The code from #137 was merged into master, so we now have a simple browser based way to update.

ThomasWaldmann, 2020-01-12:
@rexfue noted that IOTWebConf also has an Updater.

I had a look at that and the code looks quite similar, except that they do not have the "progress indication" which requires JQuery and makes troubles if you do not have an internet connect (because your laptop's wifi is connected to the esp32).

rexfue, 2020-01-12:
Nevertheless I would prefer this one, at least until we have something better. Have you looked at the solution of Feinstaubsensor? They use a server based auto update. I think one day we should use the same.

rexfue, 2020-01-12:
New OTA possibility should be mentioned in Readme !

ThomasWaldmann, 2020-01-12:
Are you saying you prefer current master branch over something based on IOTWebConf?
It looks pretty easy, so I might have that ready today (assuming the library code works).

rexfue, 2020-01-12:
No, the other way round. I prefer IotWebConfig. Also because of #140.

ThomasWaldmann, 2020-01-12:
OK, IotWebConf based updater + docs merged into master now (I reverted my previous OTA code).

So starting from V1.12.0, we have OTA capability, yay! \o/

It's still a bit simple and insecure (only protection is the esp32 AP wifi password and there is no cryptographic signature on the binary) - but better than nothing.

rexfue, 2020-01-12:
Super - thanks.

ThomasWaldmann, 2020-01-13:
Unassigning myself for now - maybe it is better to implement a good solution within IotWebConf or within Arduino framework rather than within our project. Let's see...