ecocurious2/MultiGeiger #122: alternative power supply / cabling idea

ThomasWaldmann, 2020-01-07:
primarily for @WwwBoehriDe to evaluate:

Power supply cabling based on 5.5mm/2.1mm dc connectors:

- power supply 5V 1A to dc plug: https://www.amazon.de/LiNKFOR-5-5x2-1mm-Hohlstecker-Ladeger%C3%A4t-Wandladeger%C3%A4t/dp/B07PHQMM9M/

- 10m dc extension cable: https://www.pollin.de/p/dc-verlaengerungskabel-goobay-71400-10-0-m-563066

- dc to micro usb adapter cable: https://www.amazon.de/MENGS%C2%AE-m%C3%A4nnlichen-weiblichen-Ladeger%C3%A4t-Adapter/dp/B01ER5AFM6/

I am evaluation similar parts for my geiger.

The dc connector of the adapter cable goes through a hole in the lower end cap (might seal tightly as is or can be sealed with addtl. glue / silicon).

The esp32 is right above the lower end cap.

The GM tube is above the esp32.

Note: maybe wait a bit until I have finished my geiger assembly, still working on it....

Nordrunner01, 2020-01-11:
Hmmmm..... Nope.

So aus dem Bauch raus würde ich sagen keine gute Idee.
2 Gründe:
1. Das Kabel ist nicht Wetter/Umweltfest und eigentlich nur für drinnen gedacht.
2. Bei der Kabellänge bezweiflich ich, daß am Ende noch 5V rauskommen.

Wie wäre es mit diesem Kabel für die Energieübertragung? : https://www.pollin.de/p/cat-6a-netzwerkkabel-s-ftp-outdoor-ip66-schwarz-10-m-740664

Und am Einspeisepunkt eine einstellbare Spannungsquelle um Leitungsverluste ausgleichen zu können.

Nordrunner01, 2020-01-11:
Desweiteren würde ich eine ventilierende Kabelverschraubung als Einführung ins Gehäuse vorschlagen. Etwa in dieser Art: https://www.wiska.com/de/143/cat/313/druckausgleich/

Damit hätte man 2 Fliegen mit einer Klappe geschlagen. 1. Es sieht sauber und professionell aus und 2., man hat das Problem mit der auftretenden Kondensfeuchtigkeit erledigt.

ThomasWaldmann, 2020-01-11:
Ich hab das Stromkabel / alle anderen Oeffnungen versiegelt und wuerde wohl auch etwas Trockenmittel noch in den Innenraum einwerfen. Kommt dann auf nen praktischen Versuch an, ob das oder "Ventilation" besser funktioniert.

Gibt es Schaetzwerte, wie lange so ein Kabel im Freien haelt? Wenn das ein paar Jahre haelt und dann irgendwann ersetzt werden muss, haette ich da angesichts des Preises auch kein Problem mit. Nichts haelt ewig...

WwwBoehriDe, 2020-01-12:
Kabel-Widerstand
===============
ich habe das mal durchgerechnet mit dem Kabel (reichelt.de       DC CAB 21 10M). Angenommen das ist Kupfer (was ich mal hoffe) dann komme ich auf 0,68 Ohm (beide Drähte). Somit wäre der Spannungsabfall <100 mV. Das Heltec-Board müssten wir mit einer Spannung bis zu ca. 3,6 V sicher versorgen können. Erst dann bekommt der Spannungsregler auf dem Heltec (https://www.diodes.com/assets/Datasheets/AP2112.pdf) Probleme. Also müsste man sogar 10 solche Kabel in Serie schalten können. Also daher sehe ich kein Problem

UV-Festigkeit
============
nhhh, naja, klar, das ist für Innenraum gedacht. Aber immerhin ist es schon mal schwarz und somit nicht mehr ganz so anfällig für Sonne. Allerdings haben wir das gleiche Problem mit den kompletten Gehäuse. Ich denke das würde ich einfach riskieren. Man kann ja in die Aufbauanleitung schreiben dass man auch das Kabel an stellen verlegen soll wo keine direkte Sonne hinscheint.

Zugentlastung, Belüfung
====================
Die Zugentlastung ist bisher so angedacht das man das Kabel mit Kabelbinder o. ä. an der Zwischenbodenkruke (s.
https://github.com/ecocurious/MultiGeiger/blob/master/docs/Bauanleitung/Aufbauanleitung_2019_13_09.pdf, Seite 8 )
blockiert. Ich denke das müsste auch reichen.
Die Belüftung ist so geregelt dass das ganze nach unten offen ist und so kein Wasser reinkommen kann. Ein Schutz vor Insekten ist dadurch gewährleistet dass keine Öffnung größer als 0,5 mm ist. Ich denke bisher läuft das ganz gut. Mir ist nichts gegenteiliges bekannt.

Vielleicht ist das bisher gewachsene Gehäuse-Design etwas speziell - aber man muss immer sehen dass es preiswert, einfach aufzubauen, jederzeit zugänglich (Wartungsarbeiten) und dennoch funktional sein soll. Und das sind Dinge wie komplett verschließen und Silica-Gel für den normalen Anweder eben schwierig.

ThomasWaldmann, 2020-01-16:
Power suppy: as in top post.

10m cable: as in top post.

Multigeiger V1.4 with BME280.

Measured voltage (at uC): 4.85V (display on, speaker on, wifi on, standard background radiation, Si22G tube).

Measured voltage (at power supply DC plug, no load): 4.98V

ThomasWaldmann, 2020-01-16:
Measured with better(?) DVM:

4.88V (at uC, as above)

5.02V (at power supply, no load).

ThomasWaldmann, 2020-01-16:
Same measurements, same type of power supply (bought 2 pieces):

4.92V (better DVM: 4.96V, at uC, load)
5.05V (better DVM: 5.08V, at power supply, no load)

ThomasWaldmann, 2020-02-01:
@WwwBoehriDe I see you updated the hardware assembly docs about the power supply idea.

https://github.com/ecocurious/MultiGeiger/blob/master/docs/hardware/MultiGeiger_Partlist.ods

But the parts are not on the part list yet, update the list also?

ThomasWaldmann, 2020-02-11:
@WwwBoehriDe did you see my last comment?

ThomasWaldmann, 2020-02-14:
c3a7207da7976c26f723ffa6f408cc6b2d0bf88f looks like this was fixed 3d ago.

ThomasWaldmann, 2020-02-14:
guess "all done" here, closing.

closed 2020-02-14 by ThomasWaldmann