ecocurious2/MultiGeiger #127: Frage: Ist es möglich die Platine um einen Stepdownwandler zu erweitern?

Nordrunner01, 2020-01-11:
Moin

Bezug auf: https://github.com/ecocurious/MultiGeiger/issues/122

Wenn man auf der Platine des Multigeiger einen Stepdownwandler integrieren würde, dann wäre eine Spannungsversorgung aus dem Bereich Netzwerk (PoE) möglich.
Beispielhaft wäre hier ein Wandlermodul LM2596 ( https://www.google.de/search?source=hp&ei=ClEZXpmbHdGDsAfy6Z-AAQ&q=lm2596&oq=lm2&gs_l=psy-ab.1.0.0l10.2509.7479..10119...1.0..0.128.383.3j1......0....1..gws-wiz.......0i131.Lsuflk9Qgb4 ).
Dieser ist natürlich mit seinen 3A völlig überdimensioniert und dient nur als Beispiel.

Gruß

ThomasWaldmann, 2020-01-11:
Wir verwenden ja momentan nur WiFi oder LoRa, also kein Ethernet-Kabel.

Die verwendeten EPS32-Boards haben auch (so wie sie sind) keinen Ethernet-Anschluss.

Abgesehen davon gute Idee, wenn man eh 1 Kabel verlegen muss, koennte man das natuerlich fuer Ethernet UND Power verwenden.

ThomasWaldmann, 2020-01-11:
@Nordrunner01 waer es sinnvoll, den Issue Title auf "LAN / PoE - Support?" zu aendern?

Nordrunner01, 2020-01-11:
@ThomasWaldmann 
Warum? Es geht nicht um PoE. Habe ich auch so nie geschrieben. Die Technik stammt von dort und es gibt dort entsprechende Injektoren. Es geht lediglich um die Übertragung der Betriebsspannung.

ThomasWaldmann, 2020-01-11:
BTW, auf den ESP32-Boards ist bereits ein 5V -> 3.3V Wandler/Regler drauf.

Wieviel Spielraum dadurch bei den 5V nach unten besteht waere aber zu klaeren.

Nordrunner01, 2020-01-11:
Na ja, bei den Feinstaubsensoren wird ja auch über ein 5m USBflachbandkabel der Betriebsstrom geschickt. Und es funzt.
Ich bin aber eher ein Freund von ein wenig mehr und Luft nach unten, als schon von Anfang an auf Kante zu nähen.

ThomasWaldmann, 2020-05-11:
Guess this should be answered by @rexfue and @WwwBoehriDe.

Considering the silence, I guess there is no interest to add a stepdown converter?

WwwBoehriDe, 2020-05-12:
Hi, Ich habe mal nachgeschaut: An den Modulen müsste man eine Spannung bis runter auf 4,0 V einspeisen können. Ab dann gerät der USB-Seriell-Chip "out of Spec" - was aber vermutlich unkritisch ist. Praktisch kann man vermutlich runter gehen auf 3,6 V - dann geht dem Spannungsregler die Puste aus. Aber gehen wir von 4,0 V aus dann kann 1,0 V auf der Leitung verlorengehen - bei 0,1 A. Sprich wir können 10 Ohm Leitungswiderstand erlauben. Mit unserem derzeitigen Kabel haben wir 360 mOhm pro Ader - sprich zusammen 720 mOhm (pro 10 m Stück). Sprich somit wären 130 m Kabel möglich. Allerdings könnte das mit dünnem LAN-Kabel und den zusätzlichen PoE-Komponenten ganz anders aussehen.

ThomasWaldmann, 2020-05-12:
@WwwBoehriDe The power injectors used in PoE context provide about 50V.

ThomasWaldmann, 2020-05-22:
Seems like the answer is "no" or "no interest", thus I am closing this.

I might be wrong, so just answer differently and re-open if that is the case.

closed 2020-05-22 by ThomasWaldmann