ecocurious2/MultiGeiger #136: gmc event and hv pulse correlation?

ThomasWaldmann, 2020-01-11:
When I played around with hv pulsing, I sometimes (depending on hv pulse frequency) was watching gmc pulse events correlating with hv pulses.

Sometimes this could be easily heard by a rhythmic ticking (when there was not too much ticking from actual radiation), but sometimes it was not easy to hear.

Also it is not helpful for detecting this (detecting as in "somehow obvious") that the usual background is ~1.5/s (si22g) and the hv pulse is ~1/s.

Too avoid bad measurements, we maybe should detect this error condition somehow.

ThomasWaldmann, 2020-05-10:
@rexfue if you're doing oscilloscope measurements tomorrow, could you look whether you can see that correlation?

ThomasWaldmann, 2020-05-13:
@jwoelk in case you do another round of oscilloscope measurements, could you also check this?

jwoelk, 2020-05-14:
Up to now I didn't notice any relation between tube pulses and charging pulses. But I will look again if I have some spare time. Maybe you hear the noise of coil L1?

ThomasWaldmann, 2020-05-14:
I don't think I ever noticed any noise coming from L1. But I can check again with the now slightly different tick sound (5kHz instead of 1kHz) in current code.

ThomasWaldmann, 2020-05-14:
I just "tuned" the recharging state machine a bit so it fires a lot of recharging pulses even though charging is not needed. It fires up to 100 pulses in one charge cycle, then waits only 1ms, then fires next 100 pulses. So it basically fires all the time, but the counter increments in steps of 100:

```
GEIGER:         16            9889   1.617959  0.131764      3900       16      9888  1.618123  0.131778
GEIGER:         21            9893   2.122713  0.172871      4000       37     19781  1.870482  0.152329
GEIGER:         10            7058   1.416832  0.115385      4000       47     26839  1.751183  0.142614
GEIGER:          5           12910   0.387297  0.031541      4000       52     39749  1.308209  0.106539
GEIGER:         20            8501   2.352664  0.191598      4000       72     48250  1.492228  0.121525
GEIGER:         10           11442   0.873973  0.071175      4000       82     59692  1.373718  0.111874
GEIGER:         10            9898   1.010305  0.082278      3900       92     69590  1.322029  0.107664
GEIGER:         14            9952   1.406752  0.114564      4000      106     79542  1.332629  0.108527
GEIGER:         12           10196   1.176932  0.095848      4000      118     89738  1.314939  0.107087
```

ThomasWaldmann, 2020-05-14:
OK, so there does not seem to be a correlation.

Not sure what I've seen back then when I filed this ticket, but it does not seem to be an issue now.

closed 2020-05-14 by ThomasWaldmann