ecocurious2/MultiGeiger #138: Specs/parts for the tube to mount sensor

Marwe, 2020-01-11:
I was looking for specs for the tubes to mount the sensor into and could not find a description in the Bauanleitung/DE.  Have seen the tube at 36c3 and try to grab the right thing now, would like to see specs in the docs, especially since the thickness matters for (beta) radiation shielding.

Bought some parts today and found fitting ones in the sanitary area, which can perfectly combined with the proposed tube for electric wiring:
* "Installationsrohr starr M40 3m", proposed tube
* "HT Überschiebmuffe (HTU) DN 40", straight with two seal rings
* "HT Endkappe, Muffenstopfen (HTM) DN 40" to close (on top)
* "HTEA DN 40/40 45° Einfachabzweiger", y-shaped, can be used as inlet for the wires (mount direction like an "h") and air in the center

I still need to build it, but these parts seem usable so far.

ThomasWaldmann, 2020-01-16:
I measure the tube I bought: innerdiameter 37.4mm, outer diameter 40.4mm.

But I had some slight issue with that / the pcb size, see #112.

ThomasWaldmann, 2020-05-22:
TODO: check whether there's something missing related to this in the docs.