ecocurious2/MultiGeiger #140: Settings via web GUI

rexfue, 2020-01-12:
The settings in userdefines.h should be adjustable via web interface. At least a part of it.
Started to work on that.

ThomasWaldmann, 2020-01-12:
It seems that IOTWebConf also can do that.

rexfue, 2020-01-12:
Yes, i will use it. Only the format I don't like. Maybe I can play with a litte html code.

ThomasWaldmann, 2020-02-10:
https://github.com/prampec/IotWebConf/pull/95 maybe interesting - is in latest release!

ThomasWaldmann, 2020-02-21:
Shall we move that to a later milestone?

We already have huge code changed since last release, so maybe doing a new release for public testing would be good.

rexfue, 2020-02-21:
At minimum I want to have LoRa credentials there. We have many lora fans and they ask for easy input of lora credenetials.

ThomasWaldmann, 2020-02-26:
OK, so it seems this is the last issue in next milestone.

I'll do some more cleanups while you work on it...

rexfue, 2020-02-26:
OK, I'll try to do it the next days

rexfue, 2020-02-29:
Would remove the #defines for SEAKER_TICK, LED_TICK, SEND2MAVAI ... etc. Because if we set these values in web interface, setting in user define.h is obsolete.
What do you think?

ThomasWaldmann, 2020-02-29:
@rexfue i would keep them as defaults to use as long as no web gui configuration has happened.

We need those defaults anyway and for the developers it is easier to change them as to reconfigure via web interface.

rexfue, 2020-03-02:
Defaults can be set in source anyway (by developers). And every user has to go through web interface to set his WiFi credentials - here he can set his configs too.
I decided to remove the #defines in my PR. If we really need them, we can easily bring them back.

ThomasWaldmann, 2020-03-02:
So, what's the alternative to SPEAKER_TICK? Using a literal `1` where you populate the settings form if you have no setting stored yet? I think you still need them, as defaults, as I already noted.

Also, I don't want to search the full source code to change these, they need to be at one place.

commit 59707e3e (2020-03-23): Use webpage to to set configuration, fixes #140

lorawan.cpp:
 - only compile, if CPU==STICK, i.e if we have LoRa Hardware
 - fetch LoRa credentials from config variables

multigeiger.ino:
 - use config variables instead of userdefines

transmission.cpp:
 - replace SEND2LORA by CPU==STICK
 - use config variables to decide where to send
 - send to LoRa only, if LoRa credentials are known

webconf.cpp
 - add all config variable to web page
 - do restart after saving config

userdefines-example.h
 - remove now unnecessary #defines
commit 7f85271d (2020-03-23): Merge pull request #226 from rexfue/most_settings_on_webpage

Use webpage to to set configuration, fixes #140
closed 2020-03-23 by rexfue