ecocurious2/MultiGeiger #141: duplicated changelog

ThomasWaldmann, 2020-01-12:
there is changelog information in:
- changelog.md
- multigeiger/version.h

One should be removed, it is not needed to have that at 2 places and only creates inconsistencies.
We already have them, not all versions are listed in both files.

ThomasWaldmann, 2020-01-12:
@rexfue ^^^

rexfue, 2020-01-12:
which one would you prefer?

ThomasWaldmann, 2020-01-12:
changelog.md in toplevel dir is better visible, so users can easier find and read it.

ThomasWaldmann, 2020-01-12:
please first merge my new OTA PR or it'll conflict as it updates version.h changelog.

ThomasWaldmann, 2020-01-12:
did it already. it works, i tested it.

rexfue, 2020-01-12:
Fine that it works.
keep changelog.md: OK. VERSION.MAJOR etc. still will be defined in version.h  without comments - OK?

ThomasWaldmann, 2020-01-12:
Yes.

ThomasWaldmann, 2020-01-13:
OK, I'll work on this later today.

rexfue, 2020-01-13:
I think it would be better to delete version.h completely  and write the #define VERSION_MAJOR etc. directly into multigeiger.ino

ThomasWaldmann, 2020-01-13:
ok.

commit 812bc592 (2020-01-13): move all changelog infos to changelog.md, fixes #141
closed 2020-01-13 by rexfue