ecocurious2/MultiGeiger #158: kit ideas

ThomasWaldmann, 2020-01-16:
there was some discussion today about kits and workshops:

- making kits is some effort (ordering and sorting parts, collecting money, shipping stuff).
- mechanical kit parts (plastic tube etc) are a problem due to size / shipping large sizes
- the core team can do some workshops here and there, but it does not scale and traveling costs time and money

idea:
- support workshops done by other people
- offer "kits of size N" with N = 3, 5, 10, ...

pros:
- less work than individual kits (they would not be sorted, but just contain parts for N MultiGeigers)
- including the bigger parts would be a lesser problem (tube could be roughly cut to correct length)
- could be cheaper because less work / less shipping costs?

cons:
- any?

ThomasWaldmann, 2020-03-05:
Idea from meeting: make a reduced kit by:

- make a public shopping list at reichelt for self-shopping
- links for heltec devices
- tested tubes (sometimes defect)
- exotic / hard to get components

ThomasWaldmann, 2020-05-08:
other idea: outsource
- ELV?
- Segor?
- Reichelt?
- Mitch (USA)?

Nordrunner01, 2020-05-10:
https://www.pollin.de/
https://shop.watterott.com/
https://nettigo.eu/
https://eckstein-shop.de/